Ultimaker/Cura #13911: Flipped model failure

Pascal-tgn, 2022-11-27:
### Application Version

5.4.1

### Platform

Windows 10

### Printer

Anycubic Mega i3 S

### Reproduction steps

Loaded stl file
Flipped the model
[Uploading Star_ass.zip…]()

### Actual results

Slicing failed

### Expected results

Sliced model

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Nothing else

Pascal-tgn, 2022-11-27:
Typo in ticket: software version is 5.2.1

GregValiant, 2022-11-30:
Thanks for the report.
Something happened during your upload and all I get is a link back to this page.
Could you try uploading the Project File again?

Pascal-tgn, 2022-12-01:
Hi Greg

I hope this one works.
I use 10% infill, 0.2 mm layer. Profile was taken from Anycubic Mega i3 S with only adjusted temperatures, infill and adhesion plate.
[Star_ass.zip](https://github.com/Ultimaker/Cura/files/10136081/Star_ass.zip)

MariMakes, 2022-12-02:
Hey @Pascal-tgn,

Thanks for your report 👍
Your last file also didn't really work, 
But I did some investigative journalism on Thingiverse 😉 

In the future it would be more useful to share an project .3mf file. It contains the printer and settings we need for troubleshooting. To save a project file go to File -> Save project. 
Like this [AI3MSPRO_Star_ass.zip](https://github.com/Ultimaker/Cura/files/10140290/AI3MSPRO_Star_ass.zip)

I was able to recreate your crash on my side!
I also found that if you move the model slightly away from the center of the buildplate, you are able to slice it. 
I hope that helps you get unstuck 👍 
[image]

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/13643
I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/13643

closed 2022-12-02 by MariMakes

Pascal-tgn, 2022-12-02:
Hi @MariMakes,

Sorry for issues with attachments.
I've already sorted it by slicing in Cura 4.13.1 on another computer, but your advise will be highly useful for future projects.

Thanks a lot!