Ultimaker/Cura #13913: No joy after blocking out certain areas for 100% infill

SofaKingRotten, 2022-11-27:
### Application Version

5.2.1

### Platform

Windows 10 21H2 x64, Intel Core i5-3570 CPU@3.40GHz, 16Gb RAM

### Printer

Wanhao Dupicator 6

### Reproduction steps

1. Set default infill to, say 50%. Now block out an area of the .stl you want infill to be 100%
2. Select 'per model' settings for this block, select 'modify settings for overlaps', select 'settings' and change 'Infill' to 100%
3. Slice the model and preview, the 100% infill looks correct and where it should be.
4. Print the model, but it does not infill 100% where expected, it remains 50%, whatever my default infill was set to.

### Actual results

The model is not printed with 100% infill where you designated by blocking. The whole item has the default infill everywhere.

### Expected results

I expect 100% infill where I have blocked out the model

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I have had this problem since version 4.13 when I started to try infill blocking.

closed 2022-11-27 by SofaKingRotten

GregValiant, 2022-11-27:
Thanks for the report.
Could you put together a project file ("File | Save Project") then zip it and post the sip folder here.  You didn't mention the Infill Pattern you are using and there were issues with Concentric.
Other than the problem with Concentric I know I won't be able to reproduce this behavior with my installation as I did two prints yesterday with blockers set to 100% Grid infill and there was no problem.
You will have to zoom in but this knob is 10% Grid on the top half and 100% Grid on the bottom half.  This is the actual Gcode read into AutoCad.  Cura was not involved in the translation and this is how it would print.
[image]

SofaKingRotten, 2022-11-28:
I just closed this, I tried another file and see that everything works as advertised.

GregValiant, 2022-11-28:
Thanks for getting back.