Ultimaker/Cura #13915: [Q] First layer orientation

Sineos, 2022-11-28:
### Application Version

5.2.1

### Platform

Win 10

### Printer

Ender 5

### Reproduction steps

1. Rotate a model 45° along Z
2. Observe first layer fill orientation

### Actual results

First layer fill is parallel / perpendicular to the walls

### Expected results

First layer fill 45° to the walls

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

More a question than a bug report: So far my understanding was that the first layer fill is always 45° to the walls. Apparently this is not the case if the model is rotated 45° along Z

[image]

[CE5_test.3mf.txt](https://github.com/Ultimaker/Cura/files/10103244/CE5_test.3mf.txt)

GregValiant, 2022-11-28:
Thanks for the report.
Whether it is Infill, Support, or Top/bottom layers, they have a default angle and they are always at that angle.  If the default is 45° then that is what you will see regardless of the orientation of the model.  Those features do not rotate with the model.

Within each of those settings categories are settings for "Line Directions".  So make all the settings visible, go to the Infill section and change the "Infill Line Directions".  The square brackets are required and you must use a comma as a delimiter if you enter more than one number.  Setting the line directions to [90] will rotate your infill so it is at 45° to your model.  Entering something like [45, 135] will make "Lines" infill into something that looks like a grid because it would change directions every other layer.
I'll remove the bug label and close this as what you are seeing is expected behavior and there are settings to change the angle.

Sineos, 2022-11-28:
Thanks for your reply and sorry for not being clear in the first place, but if you check the attached 3mf you will notice that `Infill Line Directions` is empty:

[image]

This means that a default of 45° shall be used.

> Those features do not rotate with the model.

I would deem this to be very counter-intuitive. A model can be rotated arbitrarily and I would think that the infill is automatically adapted. The infill direction is chosen because of robustness / stress properties and having to calculate the direction manually if I choose 53° model rotation seems not the way to go.

GregValiant, 2022-11-29:
Well, that's the way they wrote it.  It's not the only "non-intuitive" setting in there.  Part of that is because English isn't the native language of some of the developers.
"Empty brackets" does go to default and in the case of all of the "Line Direction" settings I think it's 45°.  I always use Grid and make it square to the walls when possible ( [90] ).
Sometimes it's also necessary to change the X and Y of the Infill start in order to move the infill within the model.  The settings are there, you have to practice to know the software.  It isn't the kind of app that users can pick up in three minutes and the learning curve is over.

Sineos, 2022-11-29:
Thanks for your statement and I fully understand that a slicer has a certain learning curve. Nevertheless I would consider this behavior an outright bug:
* The infill line direction is not a printer property (i.e. depending on the printer axes) but a model property
* Infill direction is essential for the mechanic properties of a part
* A single part rotated around Z can be compensated manually by setting `Infill Line Directions`
* Multiple parts on a bed may need different rotation for efficient packing --> This would lead to different line directions between the parts and there is no way to compensate. Worst case you get n "identical" parts with different mechanical properties

IMHO infill directions / angle needs to be calculated with respect to the outer wall of the model and not according to the printer's axes.

GregValiant, 2022-11-29:
_"IMHO infill directions / angle needs to be calculated with respect to the outer wall of the model and not according to the printer's axes."_
Like this?  It's one use of the Per Model settings.
I'll re-open this as you seem to feel strongly about it.  

[image]

Sineos, 2022-11-30:
The problem is the bottom layers

[image]

smartavionics, 2022-11-30:
> IMHO infill directions / angle needs to be calculated with respect to the outer wall of the model and not according to the printer's axes.

And when the outer wall is circular or wobbly, what does it do then?

Sineos, 2022-11-30:
See the screenshot above. Both are identical parts, one just being rotated around Z. The result is basically two different parts.

> And when the outer wall is circular or wobbly, what does it do then?

Of course a valid point. I'm just a user saying that this is more than an inconvenience, since we will get two different parts in the end.
Maybe take the longest vector as reference and calculate from there? Or do a linear approximation across the wobbly parts?

GregValiant, 2022-11-30:
Identical parts each rotated and each with the skins at 90° to the long outside edge.
It is called "Per Model" settings and you set the Line Directions.
[image]

Sineos, 2022-12-01:
Perfect, this was I'm looking for. Case closed. Many thanks 👍

closed 2022-12-01 by Sineos