Ultimaker/Cura #13919: Slicing failed

Vaakefjell, 2022-11-28:
### Application Version

5.1.0

### Platform

Mac OS 10.15.7 Catalina

### Printer

Creality Ender 3

### Reproduction steps

1. Created a georeferenced, textured 3D mesh of a cliff (.obj-file)
2. Edited said file in Blender, made it printable, exported to .stl
3. Imported into Cura. 
4. Slicing in standard settings (0.2mm layer height) worked fine, but fine settings (0.12mm layer height) didn't work. Minor change made to both the standard and fine profiles.

### Actual results

After a short time slicing, an error message suggesting to file bug report appears.
[cura.log](https://github.com/Ultimaker/Cura/files/10107300/cura.log)
[Failed slicing profile.curaprofile.zip](https://github.com/Ultimaker/Cura/files/10107334/Failed.slicing.profile.curaprofile.zip)

### Expected results

File to be sliced and printed as usual.

See project file in link: https://www.icloud.com/iclouddrive/03bAfjL-9dgGfg5qcT7k_dU1g#Cura_bug_report

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

https://www.icloud.com/iclouddrive/03bAfjL-9dgGfg5qcT7k_dU1g#Cura_bug_report

GregValiant, 2022-11-30:
Thanks for the report.
I don't have an Apple account so someone else will have to take a look.

MariMakes, 2022-12-01:
Hey @Vaakefjell,

Thanks for your report 👍

This mesh seems faulty
Have you tried doing a quick mesh fix with the Meshtools Plugin? 
You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools
I recommend at least fixing the "normals" to get rid of the red marks. 
If that doesn't fix it you can try a more elaborate mesh fixer like https://myminifactory.github.io/Fast-Quadric-Mesh-Simplification/.

[image]

To me, it doesn't seem like a bug in Ultimaker Cura, since the mesh is faulty and Cura is not built to alter meshes in this way, that's what CAD software is for. Do you agree that we can close the issue? 

[CE3E3V2_CE3_Skred1 - Fixed Failed slicing.zip](https://github.com/Ultimaker/Cura/files/10132841/CE3E3V2_CE3_Skred1.-.Fixed.Failed.slicing.zip)