Ultimaker/Cura #13920: non-planar 3D slicing

RaminShalom, 2022-11-29:
### Is your feature request related to a problem?

When will the non-planar 3D slicing be available as a feature in Cura?

### Describe the solution you'd like

When will the non-planar 3D slicing be available as a feature in Cura?

### Describe alternatives you've considered

When will the non-planar 3D slicing be available as a feature in Cura?

### Affected users and/or printers

When will the non-planar 3D slicing be available as a feature in Cura?

GregValiant, 2022-11-30:
_"When will the non-planar 3D slicing be available as a feature in Cura?"_
The last time I checked it looked like the process was patented.  Stepping on toes can be expensive.

FlailAway, 2022-12-02:
Correct me if I am wrong, but the idea (process) of non-planar slicing is unlikely to be Patent-able. The methods to achieve it certainly may be though. But, creating one's own coding should not step on toes.

To wit, AMD the early days -- and IBM-PC 8086 CPU hardware.

Document the approach to ensure it was not influenced by Prior Art (code), then have at it.

MariMakes, 2022-12-13:
Hey @RaminShalom,

Thanks for your report 👍
We've [all seen videos like this ]( https://www.youtube.com/watch?v=1i-1TEdByZY) and it makes us super excited for the future. 

Implementing this feature is a huuuuge undertaking in many ways, so I don't expect we can use this feature in Cura any time soon. 

In the meantime, we'll have to make do with courses like this: https://www.nonplanar.xyz/product-page/intro-to-path-design