Ultimaker/Cura #13921: Skin Edge Support - Mission Failure

PhilBaz, 2022-11-29:
### Application Version

5.2

### Platform

win 10

### Printer

Voron 2.4

### Reproduction steps

Add skin edge support layers.

### Actual results

[image]

Skin edge support is built up over the layers, but then the very last layer moves away from the top of the stack by a few line widths. It will then be an unsupported first layer and likely useless as a skin support.

I do infill at high speed 230mm+ so it will take lots of layers of skin support to produce a reliable final layer.

### Expected results

The stack should be directly under the last support layer.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[Skin Edge Fail.zip](https://github.com/Ultimaker/Cura/files/10110104/Skin.Edge.Fail.zip)
[image]

I have come to find skin edge support invaluable in getting the skin to anchor to the wall edge without missed strands. Missed strands curl and lead to visible defects in finish, they also tend to burn and leave dark spots. After bringing this up on the Voron forum, most people seem to think they are caused by nozzle gunk, but I've looked at this closely and its missed skin curling up. It snowballs into a burnt peak.

I would love to see more control over skin edge support layers. Perhaps distance from skin edge, as well as number of walls. Independent speed control to improve reliability in fewer layers. Its currently also affected by 'extra infill wall count'. In the end it would be nice to control these independently.

But most important is just getting the edge support stack to properly support itself and end under the intended target.

GregValiant, 2022-11-30:
Thanks for the report.
I'm looking at the preview on my computer and it looks a lot different (same project file I assume).
This is layer 8 and the first skin goes down on layer 9.  I'm positive that radius won't print like it looks there.  No way to hold shape around the corner so it's going to pull into a straight line.
[image]

It looks like the skin edge support goes down every other layer.  The lower layers appear to have more offset from the inner wall and then layer 8 seems pretty close to the inner wall but it does look unconnected.

BTW I think it looks better with bridging turned off.  The infill density appears to provide plenty of support for a regular skin.

PhilBaz, 2023-01-05:
@GregValiant 

Mark burton introduced a new skin support solution based on this issue. perhaps it would be of use to look at his solution for the 5.3 release. 

It solved all my issues.

GregValiant, 2023-01-05:
I'll look into it.  I use his version for Spiralize and his adaptation of Hole Horizontal Expansion is very good.

PhilBaz, 2023-01-06:
@GregValiant Great, I think it was incorporated two releases ago. Not long.

PS....in the image above, your tool paths are flat instead of cylinders. That's nice. How?

GregValiant, 2023-01-06:
I thought I'd take another look at this with since I have the new laptop.

This is with Grid infill and the line directions at [90].  The curves in the corners are much better supported and should stay where they are put. (If it were me, I'd turn off the bridge settings.  I don't see a need for them here.)
[image]

PhilBaz, 2023-01-06:
@GregValiant 

Yeah, I like it as a view option. The top 5 layers feature is nice. It would be great if it was a toggle that could be turned on and off.

PhilBaz, 2023-01-06:
@GregValiant 

That part is just a test piece. Its a chunk of a larger piece that is part of a 3 part assembly that is printed in ABS. The cubic infill is chosen for directional force in multiple axes. I believe without internal bridging the top was pillowing. All three parts have large flat surfaces on top, two are finished surfaces, one is a tolerance fit part of the assembly. 

I have done much testing with internal bridging, I have it very tuned/dialed in. It improves surface quality. At least with ABS, which is what I tend to use. 

But its interesting to see that the function works as expected with some infills but not others. The approach of stacking the support directly under the target is what I proposed to Mark, not sure what he implemented in the end, but it solved all the problems.