Ultimaker/Cura #13924: cura build problem

timmyImwu, 2022-11-29:
### Application Version

5.2.0

### Platform

windows 10

### Printer

ultimaker s5

### Reproduction steps

using the cura install instructions to build.when run the last but one- "pyinstaller ./cura_inst/Ultimaker-Cura.spec",some problems happen.

### Actual results

some folders and files can't be found. see following in detail:
88 INFO: PyInstaller: 5.6.2
88 INFO: Python: 3.10.4
699 INFO: Platform: Windows-10-10.0.19044-SP0
1294 INFO: UPX is available.
1315 WARNING: Unable to copy metadata for cura: The 'cura' distribution was not found and is required by the application
1315 WARNING: collect_data_files - skipping data collection for module 'cura' as it is not a package.
1315 WARNING: collect_dynamic_libs - skipping library collection for module 'cura' as it is not a package.
1413 INFO: Determining a mapping of distributions to packages...
5809 WARNING: Unable to determine requirements for cura: The 'cura' distribution was not found and is required by the application
5820 WARNING: Unable to copy metadata for UM: The 'UM' distribution was not found and is required by the application
5831 WARNING: collect_data_files - skipping data collection for module 'UM' as it is not a package.
5832 WARNING: collect_dynamic_libs - skipping library collection for module 'UM' as it is not a package.
5911 WARNING: Unable to determine requirements for UM: The 'UM' distribution was not found and is required by the application
5932 WARNING: Unable to copy metadata for serial: The 'serial' distribution was not found and is required by the application
5932 WARNING: collect_data_files - skipping data collection for module 'serial' as it is not a package.
5932 WARNING: collect_dynamic_libs - skipping library collection for module 'serial' as it is not a package.
6029 WARNING: Unable to determine requirements for serial: The 'serial' distribution was not found and is required by the application
6040 WARNING: Unable to copy metadata for Charon: The 'Charon' distribution was not found and is required by the application
6040 WARNING: collect_data_files - skipping data collection for module 'Charon' as it is not a package.
6040 WARNING: collect_dynamic_libs - skipping library collection for module 'Charon' as it is not a package.
6129 WARNING: Unable to determine requirements for Charon: The 'Charon' distribution was not found and is required by the application
6150 WARNING: Unable to copy metadata for sqlite3: The 'sqlite3' distribution was not found and is required by the application
7693 WARNING: Unable to determine requirements for sqlite3: The 'sqlite3' distribution was not found and is required by the application
7716 WARNING: Unable to copy metadata for trimesh: The 'trimesh' distribution was not found and is required by the application
7716 WARNING: collect_data_files - skipping data collection for module 'trimesh' as it is not a package.
7716 WARNING: collect_dynamic_libs - skipping library collection for module 'trimesh' as it is not a package.
7805 WARNING: Unable to determine requirements for trimesh: The 'trimesh' distribution was not found and is required by the application
7815 WARNING: Unable to copy metadata for win32ctypes: The 'win32ctypes' distribution was not found and is required by the application
9440 WARNING: Unable to determine requirements for win32ctypes: The 'win32ctypes' distribution was not found and is required by the application
9450 WARNING: Unable to copy metadata for PyQt6: The 'PyQt6' distribution was not found and is required by the application
9450 WARNING: collect_data_files - skipping data collection for module 'PyQt6' as it is not a package.
9450 WARNING: collect_dynamic_libs - skipping library collection for module 'PyQt6' as it is not a package.
9540 WARNING: Unable to determine requirements for PyQt6: The 'PyQt6' distribution was not found and is required by the application
9561 WARNING: Unable to copy metadata for PyQt6.QtNetwork: The 'PyQt6.QtNetwork' distribution was not found and is required by the application
9629 WARNING: collect_data_files - skipping data collection for module 'PyQt6.QtNetwork' as it is not a package.
9706 WARNING: collect_dynamic_libs - skipping library collection for module 'PyQt6.QtNetwork' as it is not a package.
9862 WARNING: Unable to determine requirements for PyQt6.QtNetwork: The 'PyQt6.QtNetwork' distribution was not found and is required by the application
9883 WARNING: Unable to copy metadata for PyQt6.sip: The 'PyQt6.sip' distribution was not found and is required by the application
9952 WARNING: collect_data_files - skipping data collection for module 'PyQt6.sip' as it is not a package.
10020 WARNING: collect_dynamic_libs - skipping library collection for module 'PyQt6.sip' as it is not a package.
10187 WARNING: Unable to determine requirements for PyQt6.sip: The 'PyQt6.sip' distribution was not found and is required by the application
10208 WARNING: Unable to copy metadata for stl: The 'stl' distribution was not found and is required by the application
10208 WARNING: collect_data_files - skipping data collection for module 'stl' as it is not a package.
10208 WARNING: collect_dynamic_libs - skipping library collection for module 'stl' as it is not a package.
10288 WARNING: Unable to determine requirements for stl: The 'stl' distribution was not found and is required by the application
10298 INFO: Extending PYTHONPATH with paths
['C:\\.conan\\b105bd\\1\\bin']
11962 INFO: checking Analysis
12069 INFO: Appending 'binaries' from .spec
12101 INFO: Appending 'datas' from .spec
13052 INFO: checking PYZ
13076 INFO: checking PKG
13078 INFO: Bootloader D:\Python\Python310\lib\site-packages\PyInstaller\bootloader\Windows-64bit\runw.exe
13078 INFO: checking EXE
13088 INFO: checking COLLECT

### Expected results

all needed folders and files should not be missing, and be found.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

no others

MariMakes, 2022-12-02:
Hey @timmyImwu,

😖 Sorry to hear you are stuck.

I'm only a fledgling member of the Build From Source Club 👑 
Before we get started, I have a few questions to narrow down the troubleshooting:

1. Can you make sure you have the latest Conan version?  
`pip install conan --upgrade `
You can check the [Conan Documentation]( https://docs.conan.io/en/latest/installation.html ) for the latest version and installation instructions

2. Did you try clearing the Conan Cache? 
`conan remove "*" -f `

3. Can you make sure you have the latest config with this command?
`conan config install https://github.com/Ultimaker/conan-config.git`

4. Can you make sure you have a default profile?
`conan profile new default --detect --force`

5. Can you make sure you have the cura-ce Artifactory repo and not the cura Artifactory repo?
Please check `cura remote list`, this will return your Artifactory Repo
If you see cura in your list, run ` conan remote remove cura`

6. With these steps you should have a clean slate and you can follow the steps on our wiki again. 
[You can find it here. ](https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source) 

If you are stuck in any way, please reach out. We are happy to help out. 
The best way to do that is to provide us with your complete run-out log. 
Not just the part that fails, but all the lines generated after you've given your command. 
We need it see your profiles and packages. 

This is a neat trick to keep our Github issues more manageable. 
[image]

I'm super curious. What do you want to achieve by building from source?
I would love to know more about your project, maybe we can help.

timmyImwu, 2022-12-03:
> Hey @timmyImwu,
>
> 😖 Sorry to hear you are stuck.
>
> I'm only a fledgling member of the Build From Source Club 👑 Before we get started, I have a few questions to narrow down the troubleshooting:
>
> 1. Can [...]

thanks, but it still fails following your instructions, I will hand in the all log later in another ticket maybe.

timmyImwu, 2022-12-03:
The following is just the entire build log:
[log.txt](https://github.com/Ultimaker/Cura/files/10145812/log.txt)

timmyImwu, 2022-12-03:
when it is built ,it seems some packages have not been included. when it runs,
 'Traceback (most recent call last):
  File "cura_app.py", line 23, in <module>
ModuleNotFoundError: No module named 'PyQt6.QtNetwork'' first happens.
the entire log is following:
[log.txt](https://github.com/Ultimaker/Cura/files/10145914/log.txt)

jellespijker, 2022-12-03:
Did you activate the virtual environment as mentioned in https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source#4-activate-the-virtual-python-environment

timmyImwu, 2022-12-05:
> Did you activate the virtual environment as mentioned in https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source#4-activate-the-virtual-python-environment before you started Cura?

we just build the cura ,not run from source.
so we use .\cura_inst\Scripts\activate_github_actions_env.ps1
                    .\cura_inst\Scripts\activate_github_actions_version_env.ps1
and the other all instructions,
but can not solve it.
you can see the all build process in the log which i have handed in.

jellespijker, 2022-12-15:
those are bash scripts specifically designed to work on a GitHub runner, where each step is a single process and environment are updated by write values to a file, which is done in those scripts.

If you're not running on a GitHub action you should replace those a
scripts with the activate as if you're running from source.

jellespijker, 2022-12-15:
Also take note of the license of Cura if you create a distribution you should make the changes available.

closed 2022-12-30 by github-actions[bot]