Ultimaker/Cura #13930: Slicing Explosion Glitch (5.2.0)

splendiferousnoctifer, 2022-11-29:
### Application Version

5.2.0 and 5.2.1

### Platform

MacOS Ventura 13.0.1, AMD Radeon Pro 5500M 4 GB

### Printer

Sovol SV01

### Reproduction steps

1. Opened Cura
2. Imported File
3. Sliced File
4. Clicked preview

### Actual results

The slicing glitches and is displayed wrongly. Looks like an explosion 
[image]
[image]

### Expected results

The slicing should be normal and look as intended.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[slicing_bug_cura.3mf.zip](https://github.com/Ultimaker/Cura/files/10115469/slicing_bug_cura.3mf.zip)
This changed suddenly from 3 days ago, where it worked completely normal.
This also occurs with any file used.

GregValiant, 2022-11-30:
Thanks for the report.
This is the "Peter Max" bug.  So far as I know it is really hard to duplicate so it is really hard to find why that happens.
This is a duplicate of #13417 and there are other reports.  My computer doesn't do it so I can't duplicate the effect.
I'll leave this one open.  That other report mentions that a re-install of Cura seemed to get rid of the problem.  You might also try going to Preferences / Configure Cura and check the setting "Force layer View Compatibility Mode..." and see if it has any effect.  Then un-check it and see if the problem still exists when you restart.

yuval2508, 2023-01-29:
> ### Application Version
> 5.2.0 and 5.2.1
>
> ### Platform
> MacOS Ventura 13.0.1, AMD Radeon Pro 5500M 4 GB
>
> ### Printer
> Sovol SV01
>
> ### Reproduction steps
> 1. Opened Cura
> 2. Imported File
> 3. Sliced File [...]

I had it too, changed material profile and it's gone. I still got "random" crashes after slicing few times, it happens in background. strange.