Ultimaker/Cura #13932: Error slicing

oseloth, 2022-11-29:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

cr-10

### Reproduction steps

using same settings as usual. Today slicing error on both version 521 and 522

### Actual results

After few seconds of slicing I get slicing error. 
[CCR10_Mish ventil v2.zip](https://github.com/Ultimaker/Cura/files/10117800/CCR10_Mish.ventil.v2.zip)

### Expected results

To be sliced

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10117813/cura.log)

oseloth, 2022-11-29:
But if I trick this model and place another one used before, it does slice it. Than I delete that old object and it still slice the new one.

GregValiant, 2022-11-30:
Thanks for the report and the file.
The model sliced for me so I can't duplicate the problem.
There have been a lot of reports regarding this behavior though.  A workaround is to rotate and/or move the model on the build plate.  When Cura comes at it from a different direction it can fool the bug.  Start with 90° CCW and see if it will slice.

I'll mark this as a duplicate and I'm going to go ahead and close it.  The problem is being worked on and should be fixed in 5.3.
There are a lot of reports of this bug.  I hope you understand.

closed 2022-11-30 by GregValiant