Ultimaker/Cura #13939: Vase mode closes bottom layer

I522I, 2022-11-30:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Artillery Genius

### Reproduction steps

1. Load Model "Mini LED tealight christmas tree" (https://www.thingiverse.com/thing:2683430)
2. selected "spiralize outer contour"
3. slice
[image]

### Actual results

Model sliced with a closed bottom layer. In prepare screen the model bottom layer is open. In normal slice Mode, the bottom layer is sliced open.
[image]

### Expected results

I can define, how many bottom layers the model should have, but cura will not close open parts. The vase mode should start after the defined bottom layers.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[AG_120mmTreeWled.zip](https://github.com/Ultimaker/Cura/files/10126361/AG_120mmTreeWled.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/10126394/cura.log)

GregValiant, 2022-11-30:
Thanks for the report.
When you enable "Spiralize"; Cura automatically enables "Remove All Holes".  The implementation of Spiralize in the 5.x versions is different than in previous versions. 
In the Mesh Fixes settings turn off remove all holes.
Because that model has a pocket in the bottom it will not print as it would have in 4.x versions of Cura.  The 5.x versions will print the pocket where the 4.x versions ignored it.  That pocket really needs to be as deep as the thickness of your bottom layers which I believe is .96 with your settings.  That way those interior walls wouldn't print and leave seams in that portion of the model.
What you see there is the new "expected behavior" in Cura 5.x.  Some people call it a regression.  It is different and needs getting used to.  I'm going to remove the bug label but I'll leave this open.

If you edit the STL and plug that hole then you can bring in a cylindrical support shape in Cura and use it as a cutting mesh and put a 30mm hole in the floor.  The cylinder would be 30 dia x .96mm thick.
I don't like to do people's homework for them but since this is a Christmas thing I'll attach my Project File.
Happy Holidays.
[GV_120mmTreeWled.zip](https://github.com/Ultimaker/Cura/files/10127083/GV_120mmTreeWled.zip)

[image]

TomMuc1, 2023-03-08:
that is still a bug. it's not christmas anymore so call it a bug in the cura-coders approach. btw this works in prusa-slicer as a no-brainer.

yurymkomarov, 2023-04-23:
> that is still a bug. it's not christmas anymore so call it a bug in the cura-coders approach. btw this works in prusa-slicer as a no-brainer.

@TomMuc1 
that is what helped me:
- disable "Remove All Holes"
- change bottom layers from 4 to 2

but yeah, it's little bit annoying )

uwezi, 2023-12-03:
> > that is still a bug. it's not christmas anymore so call it a bug in the cura-coders approach. btw this works in prusa-slicer as a no-brainer.
> 
> @TomMuc1 that is what helped me:
> 
>     * disable "Remove All Holes"
> 
>     * change bottom layers from 4 to 2
> 
> 
> but yeah, it's little bit annoying )

just experiencing the same problem/"feature" - but when I disable the "remove all holes option" I suddenly get double walls on my spiralized print, because of the finite thickness of my model's walls.