Ultimaker/Cura #13941: cura Run from source problem

timmyImwu, 2022-12-01:
### Application Version

5.2.0

### Platform

Windows 10

### Printer

Ultimaker S5

### Reproduction steps

follow the instructions from "run cura from source"

### Actual results

when executing the step:conan install . --build=missing --update -o cura:devtools=True -g VirtualPythonEnv
problem happens:

sipbuildtool/0.2.2@ultimaker/stable: WARN: Build folder is dirty, removing it: C:\.conan\ab6c18\1
sipbuildtool/0.2.2@ultimaker/stable: Copying sources to build folder
sipbuildtool/0.2.2@ultimaker/stable: Building your package in C:\.conan\ab6c18\1
sipbuildtool/0.2.2@ultimaker/stable: Generator txt created conanbuildinfo.txt
sipbuildtool/0.2.2@ultimaker/stable: ERROR: Traceback (most recent call last):
  File "D:\Python\Python310\lib\site-packages\conans\client\generators\__init__.py", line 183, in write_generators
    generator = generator_class(conanfile)
  File "D:\Python\Python310\lib\site-packages\conan\tools\microsoft\msbuilddeps.py", line 93, in __init__
    self.configuration = conanfile.settings.build_type
  File "D:\Python\Python310\lib\site-packages\conans\model\settings.py", line 279, in __getattr__
    self._check_field(field)
  File "D:\Python\Python310\lib\site-packages\conans\model\settings.py", line 275, in _check_field
    raise undefined_field(self._name, field, self.fields, self._parent_value)
conans.errors.ConanException: 'settings.build_type' doesn't exist
'settings' possible configurations are none

ERROR: Error in generator 'MSBuildDeps': 'settings.build_type' doesn't exist
'settings' possible configurations are none

### Expected results

should be ok like last three weeks when runnning first.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

no others

MariMakes, 2022-12-02:
Hey @timmyImwu,

Do you think this issue is different from: https://github.com/Ultimaker/Cura/issues/13924?
Would you like to keep this open or troubleshoot in the other ticket?

timmyImwu, 2022-12-03:
> Hey @timmyImwu,
> 
> Do you think this issue is different from: #13924? Would you like to keep this open or troubleshoot in the other ticket?

thanks， this time the building is successfull and i find the reason.
I just use the conan data before not the downloaded.

timmyImwu, 2022-12-03:
The following is just the entire build log:
[log.txt](https://github.com/Ultimaker/Cura/files/10145913/log.txt)

MariMakes, 2022-12-06:
Hey @timmyImwu,

I'll be closing this issue as a duplicate but you can continue the discussion here: https://github.com/Ultimaker/Cura/issues/13924

closed 2022-12-06 by MariMakes