Ultimaker/Cura #13942: when i try conecting to my usb via cura it says in order to moniter your print from cura please connect to printer i have the right usb

dfevrgetrgewgr, 2022-12-01:
### Application Version

5.2.1

### Platform

windows 11

### Printer

ender 3 pro

### Reproduction steps

i took i file from thangs and put it in cura then i enserted the usb in the printer and my laptop

### Actual results

i doesnt give me the option to print from usb

### Expected results

i expected to get the choice to print from usb

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

that was it

GregValiant, 2022-12-01:
Thanks for the report.
Since no Ultimaker printers use USB Printing it is unmaintained by the Ultimaker Cura team. It either works or it doesn't depending on each user.  UltiMaker isn't going to put time and effort (money) into something they don't use.  It is still included in Cura for folks who want to use it, but there is so much different hardware out there that some is bound to be incompatible.  Cura is open source software so any developer could update USB Printing.  I notice no one has.

There are some things you can check:
Go to the Marketplace and click on the "Gear" icon.  Scroll down and make sure "USB Printing" is installed.
With the USB cord plugged in to the laptop and the printer - go to Settings on the computer and then "Device Manager".  Under "Ports (COM and LPT)" there should be a line like "USB-SERIAL CH340(COMxx) where the xx is the number of the com port.  If it isn't there (or if "Ports (COM and LPT) isn't there) then the driver isn't installed OR you can try using a different USB.  Sometimes a driver will be installed on a particular USB port rather than all of them.  That is a function of the operating system.

I don't have any problem communicating with my own Ender 3 Pro via the USB (although I don't print over it because USB printing is "iffy" at best).  People seem to like the Raspberry PI with Octoprint combination.  I'm good with the SD card.  It's simple and easy to understand.  I did add an adaptor for a regular size SD card so I don't have to handle the little one.  I kept losing it.

I'll leave the bug label on this but realistically it's something you may not get to work.

dfevrgetrgewgr, 2022-12-01:
> Wich gear icon in the marketplace?

GregValiant, 2022-12-01:
There is only one.  The plugins are in alphabetical order.
[image]