Ultimaker/Cura #13944: Non-horizontal only slicing

FalcoGer, 2022-12-01:
### Is your feature request related to a problem?

When printing large overhangs, a lot of support is usually needed because there is no material beneath the nozzle. Removing support is annoying, produces waste and uses up filament that is better spent for more prints.

When slicing, Cura slices in horizontal layers only and doesn't fully take advantage that a 3d printer can move in all 3 axis. If it were to slice differently, overhangs would have support by parts already printed to the side of them.

### Describe the solution you'd like

When slicing, the part could be sliced in cones, starting from the origin, or perhaps a user defined point. The cone angle would be determined by the physical layout of the print head to avoid bumping the part. A relatively shallow cone angle would already result in making overhangs a lot more stable to print since there is already a higher part to the side of the newly printed part that it can rest on before it cools.
The positive to this approach is that it should be relatively simple to implement by generating the slices that intersect with a cone surface.
The downside to this approach is that it only works for overhangs radially out from the center point.

### Describe alternatives you've considered

Overhangs could be detected and then then slicing happens at an angle leading up to the overhang to provide support to it. The slices would have to be curved to accommodate this. A maximum angle also needs to be provided here due to the physical construction of the print head.

Layers would be curved in complex ways, depending on the geometry of the object. The curve would gradually increase up to the maximum angle where the overhead begins.

The upside to this approach is that it works for any overhang in any direction. It also means that horizontal surfaces are in one layer. It also means that the z-axis is used less during printing, reducing wear.

The downside is that layer width is adaptive and it seems to me a lot more complicated to implement. Complicated shapes could be a problem to compute.

### Affected users and/or printers

Everybody with a 3 linear axis printer.

### Additional information & file uploads

Simple cone slicing, cross section view, slices form the surface of a cone. 

[image]

Dynamic angle slicing, exaggerated, also slices should be curved, not abruptly changing angle.
[image]

Refs:
https://www.youtube.com/watch?v=1i-1TEdByZY
https://www.cnckitchen.com/blog/guide-how-to-use-conical-slicing

MariMakes, 2022-12-13:
Hey @FalcoGer,

Thanks for your report 👍

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/13920
Implementing this feature is a huuuuge undertaking in many ways, so I don't expect we can use this feature in Cura any time soon.

Shockah, 2023-08-05:
It seems the referenced duplicated issue is now gone. I'm not sure if it included any additional information or not, but at the very least this should no longer be marked as duplicate?