Ultimaker/Cura #13946: Rectangular model with penetrating features and an extra layer on the inner wall

CHAKone, 2022-12-01:
### Application Version

5.2.1

### Platform

windows10

### Printer

vyper

### Reproduction steps

[image]
[image]
[image]
[image]
[image]

### Actual results

As shown in the figure, if the rectangular inner wall has a through (penetration) feature in the model, the slice will appear to have an extra layer of inner wall at the location of the (penetration) feature. The problem is still the same in test version 4.13.

### Expected results

[image]
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

This problem occurs for all rectangles with (through) or through features

GregValiant, 2022-12-01:
Thanks for the report.
You didn't include a project file so I will guess and say that you have "Hole Horizontal Expansion" enabled.  When Cura slices where those slots are then the hole isn't a hole anymore but some other shape.  Once it gets above the slots then it's a hole again and Hole Horizontal Expansion kicks back in and you get that ledge.  You will need to turn off Hole Horizontal Expansion to get rid of that step.
The @smartavionics fork of Cura (based on the 4.x versions of Cura) has a setting for this condition.

I'll remove the bug label as this is (for now) the expected behavior of UM Cura.  I hope you understand.

CHAKone, 2022-12-02:
> 
Thanks for your reply to answer your question. After changing the settings as you suggested, the slice preview is now normal.
Thank you

GregValiant, 2022-12-03:
I'm going to close this.  Let me know if you think that is premature.

closed 2022-12-03 by GregValiant