Ultimaker/Cura #13947: how to remove ultimaker printer outside cura ?

j-snijder, 2022-12-01:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

Ultimaker Original dual extrusion

### Reproduction steps

Opened a 3mf file from thingiverse and it added this printer, which I do not own, but is embedded in the 3mf file.

### Actual results

Now Cura is fully frozen and will not do anything.

Is Ultimaker doing this on purpose to dump users of their programm which use other nrand printers :) ?

### Expected results

Cura to run normally.

HOW CAN I remove this printer without opening cura, is there a config file I can edit ?

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

HOW CAN I remove this printer without opening cura, is there a config file I can edit ?

MariMakes, 2022-12-02:
Hey @j-snijder,

There is a way to do this.  Your personal configuration can be found by copy pasting this in your file explorer: 
%APPDATA%\cura\5.2 

References to this introduced printer probably live in the folders User, Quality Changes, Machine Instances, Extruders, and Definition changes. But you'll have to be pretty precise or you'll get a corrupted configuration message. 

If you get the corrupted configuration message, don't worry with the start-up optimizer plugin the files you have forgotten will also be removed. 
You can download it here. https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/StartOptimiser

Hope this helps. Good luck! 💪

j-snijder, 2022-12-02:
Hi :)

Sorry, I have abandened v 5.2, too much of a hassle,perhaps if the problems are solved I'll rethink but for now I am sticking with v 5.1.1 under windows 11 because thats running fine on my machine (tested it enough to go there)

Good luck with finding the problem.

j-snijder, 2022-12-02:
Sorry but I have abandoned v 5.2, too much hassle for me to handle :(

Running v 5.1.1 on windows 11 now and no problems there.
Perhapps if in the future v 5.2.x gets stable I'll rethink my decision but for now it 5.1.1 for me and for really fast and reliable use off to creality slicer.

Good luck with finding the bugs in 5.2.x

MariMakes, 2022-12-02:
Thanks for the update. I'll close the issue.

closed 2022-12-02 by MariMakes