Ultimaker/Cura #13948: Slicing Failed

NeuroKuma, 2022-12-01:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

Double click on STL file (Flexi Crystal Dragon from Cults3D - FlexiCrystalDragon.stl) to open in Cura.  Click Slice to begin generating the GCode file.

### Actual results

About 10-15% into slicing the file, progress stops and the error "Slicing Failed" pops up (see attached image).
[image]

### Expected results

Slicing completes and GCode file is generated.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/10136820/cura.log)
[CE3E3V2_FlexiCrystalDragon.zip](https://github.com/Ultimaker/Cura/files/10136831/CE3E3V2_FlexiCrystalDragon.zip)

GregValiant, 2022-12-02:
Thanks for the report.
That dragon has showed up here before.  It was so full of errors it simply couldn't be fixed.  This version is much better but it still has something wrong with it.
After I rotated it on the build plate in Cura it sliced.  It took a while and I sliced it without support the first try.  When I turned Generate Support on it wouldn't slice and I had to rotate a couple of more times and then it did slice.
This behavior has been noted and the Cura team is collecting models to help find the bug.  I'll mark this as a duplicate and someone from the Cura team will take a look.
@MariMakes This model doesn't appear to have errors but the repair utilities I use spent a lot of time on it.  Mesh Mixer crashed trying to fix it and formware.co/OnLineSTLRepair also reported no errors and then produced this report.  
--> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
--> 0 Non-manifold edges (?)
--> 0 Inverted faces (?)
--> 0 Degenerate faces (?)
--> 0 Duplicate faces (?)
--> 0 Disjoint shells (?)
-> Repairing: 100.00%
----- Repair completed in 210985ms ------
-> **Vertex count changed from 91423 to 171974 (+80551)**
-> **Triangle count changed from 195492 to 418460 (+222968)**
So I think although it finally did slice, there is something going on with that model.  The repaired model also came back as "not watertight" so something is off.
[image]

MariMakes, 2022-12-02:
Hey @NeuroKuma,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

Thats a cool model. You can indeed only slice it if you move it a little away from the center. 
[image]

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/13643
We are indeed investigating this behavior and this is a great addition!

closed 2023-01-27 by casperlamboo