Ultimaker/Cura #13950: Model Not Slicing

TheHackerArtist, 2022-12-02:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Custom

### Reproduction steps

Import model
Slice mode
Got error : Slicing failed with an unexpected error
[cura.log](https://github.com/Ultimaker/Cura/files/10138057/cura.log)
[Violin-Big-Body.zip](https://github.com/Ultimaker/Cura/files/10138059/Violin-Big-Body.zip)

### Actual results

No slicing :(

### Expected results

Sliced model

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Not really

TheHackerArtist, 2022-12-02:
I changed the Slicing Tolerance feature under experimental tab to "Middle" instead of "Exclusive" and also changed the Minimum Polygon Circumference feature to 1.0 (4.0 at time of crash). I change both setting then resliced so cant speak to which was responsible. But I think that an interesting path to understand where the bug may have been.

Cheers!

TheHackerArtist, 2022-12-02:
Update: Only "Exclusive" slicing tolerance fails. Middle and inclusive both work and the min poly circumference doesnt cause the crash 1.0 and 4.0 both work fine with Middle slicing tolerance.
Cheers

MariMakes, 2022-12-02:
Hey @TheHackerArtist,

Thanks for your report 👍

I can recreate your slicing engine crash.
I also believe it's related to one of the changed settings,

[image]

I'll bring it up to the team to take a look!

jellespijker, 2024-01-14:
Thank you for reporting this issue, I can confirm that the project file still crashes with our latest internal build 5.7.0-alpha.0+3abb55 . For internal reference the crash is logged crash under Sentry issue: [CURAENGINE-20](https://ultimaker-o7.sentry.io/issues/4863879749/)