Ultimaker/Cura #13956: Application does not show in Mac OS Launchpad after install

johncarlson21, 2022-12-02:
### Application Version

5.2.1

### Platform

Mac OS Monterey 

### Printer

Ender 3V2

### Reproduction steps

Install App
Look in Launchpad for Cura

### Actual results

Does not show in Launchpad

### Expected results

Show in Launchpad

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I also have Cura 4.12 installed.

MariMakes, 2022-12-06:
Hey @johncarlson21,

Thanks for your report 👍

Personally I'm a Windows person so I had to ask around what a Launchpad is. 
I have some questions to try to understand it better before I bring it up to the team. 

Do I understand correctly that you have 2 versions of Cura installed, 4.12 and 5.2.1.
Does only one of them shows up in your launch pad? 
Does this stop you from opening Cura at all, or just the last version? 
Do you have other software that has several versions installed in parallel? 
How are those displayed?

MariMakes, 2022-12-06:
Hey @johncarlson21,

Thanks for the answers. 
So odd that it used to work for 5.0 🤔 

I'll bring it up to the team to see what they can do to improve it. But I expect that if it's not a quick fix that it will not get a high priority to implement. 
Fingers crossed 🤞

johncarlson21, 2022-12-06:
It looks like somehow the UI changed between the 5.0.x and 5.2.1

MariMakes, 2022-12-09:
Hey @johncarlson21,

Thanks again for the report.
I brought it up to the team and they figured out that we are missing a file info.plist in the build. 

We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-10007

Thanks for the report! 👍

johncarlson21, 2022-12-09:
That is great

drfaizal, 2023-06-25:
any updates on this problem?

MariMakes, 2023-07-11:
Hey @drfaizal,

Thanks for the bump 🤛 

Since Cura 5.3.0 we've made PKGs available for download. They seem to resolve this issue. 
But I cannot confirm, since I'm a Windows user. 

You can download the PKG for the latest version of Cura here: 
https://github.com/Ultimaker/Cura/releases/tag/5.4.0

Can you let us know if that resolves your issue?

closed 2023-07-25 by github-actions[bot]