Ultimaker/Cura #13959: won't slice

benzdoc, 2022-12-03:
### Application Version

5.2.1

### Platform

IOS 13.0.1

### Printer

reality ender 3 pro

### Reproduction steps

file will slice and print at it's original size. if increase by 50% slicing crashes. I have run the mesh tool and it says the model is watertight.

### Actual results

[image]

### Expected results

never had a slice fail

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

won't let me load project file, says it is not supported,  original stl are also not supported. Link to file... https://www.thingiverse.com/thing:2286867

GregValiant, 2022-12-03:
Thanks for the report.
I didn't have any trouble slicing it so I'm unable to duplicate the problem.
There is a bug that can keep models from slicing.  Try rotating and/or moving the model on the build plate.  The bug can often be fooled by changing the orientation or location of the model.
I'll mark this as a duplicate and someone from the Cura team will take note.

[image]

benzdoc, 2022-12-03:
Putting it in the home corner and rotating it 90 degrees worked. Thanks for the suggestion.

O L (Chip) Hughes
1967 MB 230SL
1958 MB 190
Henrico, Virginia
***@***.***