Ultimaker/Cura #13960: Can't generate supports

aleff616, 2022-12-03:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

Ender 3 v2

### Reproduction steps

Slice with tree Supports
[image]
[image]

### Actual results

Error at slicing 

[image]

### Expected results

Slice and generation of supports

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Project file

https://mega.nz/file/mxIyiRTK#g07Fa_eA37imj8gv5UYn2biZ0Ggj088e-jeEKWxbQMc

GregValiant, 2022-12-03:
Thanks for the report and the file.
I was able to get the model to slice after rotating it on the build plate.
I'll mark this as a duplicate as there have been numerous reports of this bug.  Try moving and/or rotating the model on the Cura build plate.  I rotated it 90° CW and it sliced.

@MariMakes you may want to add this one to the list.

I exported the model as an STL to check for errors and at 234mb and 2.1 million vertices and 0.5 million triangles it is the highest resolution model I've sliced.
I would suggest changing the Overhang Support Angle to 60° since at 45° a LOT of support is being generated and its a 49 hour print.

[image]

MariMakes, 2022-12-05:
Hey @aleff616,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

@GregValiant is right. This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/13643

jellespijker, 2023-12-10:
Still crashes in 5.7.0-alpha