Ultimaker/Cura #13961: immediate crash upon start

dotysan, 2022-12-03:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

unable to

### Reproduction steps

Run program.

### Actual results

Dump Summary
------------
Dump File:	Ultimaker-Cura.exe.8160.dmp : C:\Users\Curtis\AppData\Local\CrashDumps\Ultimaker-Cura.exe.8160.dmp
Last Write Time:	12/3/2022 9:05:47 AM
Process Name:	Ultimaker-Cura.exe : C:\Program Files\Ultimaker Cura 5.2.1\Ultimaker-Cura.exe
Process Architecture:	x64
Exception Code:	0xC0000409
Exception Information:	
Heap Information:	Not Present

System Information
------------------
OS Version:	10.0.19045
CLR Version(s):	

Modules
-------
Module Name	Module Path	Module Version
-----------	-----------	--------------
Ultimaker-Cura.exe	C:\Program Files\Ultimaker Cura 5.2.1\Ultimaker-Cura.exe	0.0.0.0
ntdll.dll	C:\Windows\System32\ntdll.dll	10.0.19041.2130
kernel32.dll	C:\Windows\System32\kernel32.dll	10.0.19041.2251
KERNELBASE.dll	C:\Windows\System32\KERNELBASE.dll	10.0.19041.2193
user32.dll	C:\Windows\System32\user32.dll	10.0.19041.2130
win32u.dll	C:\Windows\System32\win32u.dll	10.0.19041.2251
gdi32.dll	C:\Windows\System32\gdi32.dll	10.0.19041.2130
gdi32full.dll	C:\Windows\System32\gdi32full.dll	10.0.19041.2130
msvcp_win.dll	C:\Windows\System32\msvcp_win.dll	10.0.19041.789
ucrtbase.dll	C:\Windows\System32\ucrtbase.dll	10.0.19041.789
advapi32.dll	C:\Windows\System32\advapi32.dll	10.0.19041.2130
msvcrt.dll	C:\Windows\System32\msvcrt.dll	7.0.19041.546
sechost.dll	C:\Windows\System32\sechost.dll	10.0.19041.1865
rpcrt4.dll	C:\Windows\System32\rpcrt4.dll	10.0.19041.1806
comctl32.dll	C:\Windows\WinSxS\amd64_microsoft.windows.common-controls_6595b64144ccf1df_6.0.19041.1110_none_60b5254171f9507e\comctl32.dll	6.10.19041.1110
imm32.dll	C:\Windows\System32\imm32.dll	10.0.19041.2193
python310.dll	C:\Program Files\Ultimaker Cura 5.2.1\python310.dll	3.10.4150.1013
ws2_32.dll	C:\Windows\System32\ws2_32.dll	10.0.19041.546
version.dll	C:\Windows\System32\version.dll	10.0.19041.546
VCRUNTIME140.dll	C:\Program Files\Ultimaker Cura 5.2.1\VCRUNTIME140.dll	14.32.31326.0
cryptsp.dll	C:\Windows\System32\cryptsp.dll	10.0.19041.546
rsaenh.dll	C:\Windows\System32\rsaenh.dll	10.0.19041.1052
bcrypt.dll	C:\Windows\System32\bcrypt.dll	10.0.19041.1023
CRYPTBASE.dll	C:\Windows\System32\CRYPTBASE.dll	10.0.19041.546
bcryptPrimitives.dll	C:\Windows\System32\bcryptPrimitives.dll	10.0.19041.1415
python3.dll	C:\Program Files\Ultimaker Cura 5.2.1\python3.dll	3.10.4150.1013
_ctypes.pyd	C:\Program Files\Ultimaker Cura 5.2.1\_ctypes.pyd	3.10.4150.1013
ole32.dll	C:\Windows\System32\ole32.dll	10.0.19041.1202
combase.dll	C:\Windows\System32\combase.dll	10.0.19041.2130
oleaut32.dll	C:\Windows\System32\oleaut32.dll	10.0.19041.985
_bz2.pyd	C:\Program Files\Ultimaker Cura 5.2.1\_bz2.pyd	3.10.4150.1013
_lzma.pyd	C:\Program Files\Ultimaker Cura 5.2.1\_lzma.pyd	3.10.4150.1013
_socket.pyd	C:\Program Files\Ultimaker Cura 5.2.1\_socket.pyd	3.10.4150.1013
[... 135 lines cut ...]
shlwapi.dll	C:\Windows\System32\shlwapi.dll	10.0.19041.2075
profapi.dll	C:\Windows\System32\profapi.dll	10.0.19041.844
qwindows.dll	C:\Program Files\Ultimaker Cura 5.2.1\PyQt6\Qt6\plugins\platforms\qwindows.dll	6.2.4.0
wtsapi32.dll	C:\Windows\System32\wtsapi32.dll	10.0.19041.546
powrprof.dll	C:\Windows\System32\powrprof.dll	10.0.19041.546
umpdc.dll	C:\Windows\System32\umpdc.dll	0.0.0.0
qwindowsvistastyle.dll	C:\Program Files\Ultimaker Cura 5.2.1\PyQt6\Qt6\plugins\styles\qwindowsvistastyle.dll	6.2.4.0
opengl32.dll	C:\Windows\System32\opengl32.dll	10.0.19041.2193
glu32.dll	C:\Windows\System32\glu32.dll	10.0.19041.2193
clbcatq.dll	C:\Windows\System32\clbcatq.dll	2001.12.10941.16384
AppXDeploymentClient.dll	C:\Windows\System32\AppXDeploymentClient.dll	10.0.19041.2075
ig9icd64.dll	C:\Windows\System32\DriverStore\FileRepository\ki123273.inf_amd64_417b6a6ec9834bf2\ig9icd64.dll	22.20.16.4708

### Expected results

Anything.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Reinstalled twice & rebooted. No change.

MariMakes, 2022-12-05:
Hey @dotysan,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

Your issue reminds me a bit of https://github.com/Ultimaker/Cura/issues/13728
Is it possible that your cache is full? Can you try deleting the C:\Users<username>\AppData\Local\cura<version>\cache folder ?

If that doesn't help, let us know and we'll try something else 👍

GregValiant, 2022-12-05:
@MariMakes you're starting to sound like my doctor.

dotysan, 2022-12-23:
Sorry for the late reply. The folder C:\Users\Curtis\AppData\Local\cura\5.2\cache was created, but is empty. So the crash must have happened just after the cache folder was created, but before any UI or other actions could be done.

I just installed the alpha+xmas release and same symptom. Crash and empty C:\Users\Curtis\AppData\Local\cura\5.3\cache.

dotysan, 2022-12-25:
So I can run Cura from source just fine on this PC! But the installed binary still crashes.

Here are its dying gasps:
[image]

MariMakes, 2022-12-29:
Hey @dotysan,

Thanks for the update.
I'll bring the new insights up to the team.
Perhaps they can see some hints in your screenshot.

dotysan, 2022-12-30:
Hi @MariMakes, happy to help you reproduce, if you have ideas how...