Ultimaker/Cura #13962: Slicing Failed

feesta, 2022-12-03:
### Application Version

5.2.1

### Platform

MacOS

### Printer

Ender 3

### Reproduction steps

1. Load model
2. Click "Slice"

### Actual results

Error pops up that slicing failed and recommended to file a bug.

### Expected results

Expected to slice or to describe the problem so it can be addressed.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

cura.log
[cura.log](https://github.com/Ultimaker/Cura/files/10147501/cura.log)
[CE3PRO_otis7-2-Otis.3mf.zip](https://github.com/Ultimaker/Cura/files/10147503/CE3PRO_otis7-2-Otis.3mf.zip)

feesta, 2022-12-03:
Running "Fix Model Normals" from Mesh Tools made the model slicable. Maybe the issue should be caught and prompt to fix the model. Even better would be to give a shortcut to the fix the normals or at least state that that's a recommended step to troubleshoot.

MariMakes, 2022-12-06:
Hey @feesta,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

Your mesh indeed seems faulty. I'm getting this notification from the Meshtools Plugin.
[image]

I've also noticed that the model slices without making model fixes if you move the model slightly away from the center of the buildplate. 
[image]

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/13643
We have a ticket on our backlog with the intent to improve this behavior.
I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/13643

Thanks again for your report 😊

closed 2022-12-06 by MariMakes