Ultimaker/Cura #13970: slicing failed

r2n23d, 2022-12-05:
### Application Version

5.2.1

### Platform

Windows 11 Home

### Printer

creality cr-6 se

### Reproduction steps

I imported a file that I created from tinkercard and made my adjustments on cura and went to slice and it came up with this error code
[image]

### Actual results

nothing happens, I click on the button to report bug and I can continue makes changes, just cant slice to then save to sd card to print. File should have sliced to Gcode but did not work

### Expected results

cura should have sliced the file to a gcode so I could then export it to my sd card.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Zoey Ornament.zip](https://github.com/Ultimaker/Cura/files/10149944/Zoey.Ornament.zip)

r2n23d, 2022-12-05:
I could clear the build plate and import a file that has been successful and it worked but not this particular file.

MariMakes, 2022-12-06:
Hey @r2n23d,

Thanks for your report 👍

This mesh seems faulty
Have you tried doing a quick mesh fix with the Meshtools Plugin? 
You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools

I suggest to check your file in your modeling program again because I expect that these parts are not going to print correctly
[image]
[image]

To me, it doesn't seem like a bug in Ultimaker Cura, since the mesh is faulty and Cura is not built to alter meshes in this way, that's what CAD software is for. Would you still like to keep this issue open or do you agree that we can close it?

r2n23d, 2022-12-06:
Thank you for your help. Yes we can close it. I will try fixing the file.

Sent from Nikki's iPhone

[image]<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F40423138%2F205917113-231b5993-d65a-4a7b-8ab8-d2cd84bd6bb5.png&data=05%7C01%7C%7Cb09a4c6bee05448a2db908dad788a259%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C638059279091150956%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=qh0t3YRPNZBA8gvMHp0fKXv%2BREiDZYVbc5agfkF97PM%3D&reserved=0>
[image]<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F40423138%2F205917335-5053f925-6662-4c45-9213-c5b6a819af0d.png&data=05%7C01%7C%7Cb09a4c6bee05448a2db908dad788a259%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C638059279091150956%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=xuAUk7zwklNYZKo2PToB14WB0UxF0DIkeHvuUXnAbiA%3D&reserved=0>