Ultimaker/Cura #13976: Cura v5.2.1 - forever slicing

pportnawak, 2022-12-05:
### Application Version

5.2.1

### Platform

Debian 11

### Printer

Monoprice Mini Delta V2

### Reproduction steps

Change settings for support

### Actual results

slicing ....

### Expected results

preview of stl

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[alfbug.zip](https://github.com/Ultimaker/Cura/files/10156288/alfbug.zip)

GregValiant, 2022-12-05:
Thanks for the report.
This is a shot from Mesh Mixer.  The red balls indicate errors in the model.  They can have an effect on the slice and the model might not print correctly.
[image]

This is Mexh Mixer's attempt at repair.  It looks like the repair was incomplete and there are holes in the model and Cura can't slice this as the model isn't solid.
[image]

I tried MS 3D Builder and it's repair finally got it to slice.

This might be a Cura bug but your model has some serious issues that are at least making the water muddy so it's hard to tell what is really going on.

Here is the repaired model sliced.  I had to remove the blockers because I replaced the model.
[image]

pportnawak, 2022-12-06:
ok.
the original file (stl) didn't have holes. I opened it again and there it is.
IMHO  if there are holes in the model, there should be a warning somewhere in cura.
[image]

closed 2022-12-06 by pportnawak

GregValiant, 2022-12-06:
If you would go to the MarketPlace there is a very handy plugin called "Mesh Tools" that (when loaded) does perform a check when you load a model.  It also has some repair capability that is fine for small defects.  Your model needed some serious repair though.

pportnawak, 2022-12-06:
Thank you for pointing me to the plugin.
The stl i loaded is - and was - whole as meshlab saw it.The export didn't and i don't know why.My first print was okay besides having no support.
regards