Ultimaker/Cura #13979: Cura 5.2.1 Cannot select material with Ultimaker 2

Kastimazillion, 2022-12-06:
### Application Version

5.2.1

### Platform

win 11

### Printer

Ultimaker 2, Ultimaker 2+

### Reproduction steps

No options for any materials when UM2 is selected as printer, 
Options to create or import profiles greyed out. 
Useless. 

Changing to  UM2+ allows expected material selection, but I don't have a 2+.

I realize it's EOL hardware, but seriously. 

[image]

### Actual results

Frustrated, search web for "Cura 5.2.1 material import" to try to import my Cura 4.0 basic ass ABS profile. 
Fail, no hope.

File bug to see if "By Design".

### Expected results

Allow material selection.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

O_0

MariMakes, 2022-12-06:
Hey @Kastimazillion,

Thanks for your report 👍

This is indeed By Design.
If I understand correctly this is not the case because it's End of Service but because the material flows were different back than. 
You can read more about it here. https://support.makerbot.com/s/article/1667411304549

To me, it doesn't seem like a bug in Ultimaker Cura, since we have a solution in place. 
I'll be closing this issue. I hope you understand.

closed 2022-12-06 by MariMakes

Kastimazillion, 2022-12-06:
@MariMakes I'm not trying to import a special material profile, just ANY material profile so I can use this version of Cura.
**Just to be clear, you are saying it is by design that after selecting Ultimaker 2 as the printer, there are no options for materials available, generic, Ultimaker commercial or otherwise rendering Cura 5.2.1 uselsess for Ultimaker 2 users.**

fieldOfView, 2022-12-07:
> Just to be clear, you are saying it is by design

**Yes**. For the Ultimaker 2, materials are handled on the printer, not in Cura. The idea was to be able to slice once and print with any material. Since then we have all come to the realization that it does not work like that; you need to slice differently for different materials.

You can change how Cura handles this by changing the firmware flavor from "Ultimaker 2" to "Marlin" in the Machine Settings. That way you get materials in Cura and what you select in Cura overrides what you select on the printer. Your printer will inform you about that when you start a print. As a side-effect you gain the ability to use printing over USB, though that is still not a recommended thing to do.

Kastimazillion, 2022-12-07:
Thanks @fieldOfView, I'll try that. 
It wasn't clear that it was being done that way. Previous versions of Cura allowed me to select a material. I thought the onboard settings were only used for material changes. I can't believe I've been printing with this for so long and didn't know that.  ¯\_(ツ)_/¯