Ultimaker/Cura #13980: Cura 5.2.1 fails to slice

Anathema-Device, 2022-12-06:
### Application Version

5.2.1

### Platform

Windows 11 Home

### Printer

Voron 0.1

### Reproduction steps

Just try to slice the tension ring

### Actual results

slicing fails with a message to report a bug.  Closing and restarting Cura did not help

### Expected results

Slicing to complete successfully

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[V0_MotorBeltTension_Ring-FAILS.zip](https://github.com/Ultimaker/Cura/files/10162044/V0_MotorBeltTension_Ring-FAILS.zip)

GregValiant, 2022-12-06:
Thanks for the report and the project file.
With the Cura plugin "Mesh Tools" loaded you would get this warning when opening the model file.
[image]

Here is the view in Mesh Mixer showing the problem areas.
[image]

The model wouldn't slice as received.  I rotated your model (without any repair) on the Cura build plate and it sliced.
This may be a duplicate (I'm going to mark it as one) of several other reports where slicing failed because of the orientation of the model on the build plate.  It is hard to tell if it is that bug is the problem because the model has errors and those errors are likely having an effect.
The Cura team may want to look at this report.

Here is a report on your model (from formware.co/onlineSTLrepair).  Both Mesh Mixer and MS 3D Builder were able to repair the model.

-> Reading file and indexing vertices
-> Analyzed your file:
--> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
--> 495 Non-manifold edges (?)
--> 3859 Inverted faces (?)
--> 560 Degenerate faces (?)
--> 17 Duplicate faces (?)
--> 0 Disjoint shells (?)

Anathema-Device, 2022-12-06:
Interesting.  That would appear to be the issue.  I opened the stl in the 3d builder on Windows and let it repair the file then Cura could slice it just fine.

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release
[image]

closed 2024-01-14 by jellespijker