Ultimaker/Cura #13986: Error Slicing

WWGG12, 2022-12-07:
### Application Version

5.2.1

### Platform

OSX

### Printer

Creality CR X

### Reproduction steps

1. Opened STL
2. Slice
3. 
[HomepodMini.3mf.zip](https://github.com/Ultimaker/Cura/files/10176865/HomepodMini.3mf.zip)

### Actual results

Error

I reported this error before for 5.0.0
in 4.17 it does slice perfectly

### Expected results

Sliced file

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

cannot find log files

GregValiant, 2022-12-07:
Thanks for the report and the project file.
I can duplicate this with your project file.  The model has some errors but even after repairing it I couldn't get it to slice.  PrusaSlicer took a long time but finished the slice. 

Just for curiosity sake - this image shows the errors in Mesh Mixer.
[image]

Someone from the Cura team will take a look.

GregValiant, 2022-12-07:
@MariMakes I added a 0.4 x 3 x 80 cutting mesh to the model and this project will slice.  
[GV HomepodMini with Cutting Mesh.zip](https://github.com/Ultimaker/Cura/files/10178075/GV.HomepodMini.with.Cutting.Mesh.zip)

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release

[image]

closed 2024-01-14 by jellespijker