Ultimaker/Cura #13988: slicing failed

Arkirianne, 2022-12-07:
### Application Version

5.2.1

### Platform

windows 10

### Printer

ender 3 pro

### Reproduction steps

1. Drag & drop 3D model
2. Click Slice
3. Cura tell slicing failed

### Actual results

I can't print this model

### Expected results

Print the model

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

no

Arkirianne, 2022-12-07:
> Thanks for the report.
> Your project sliced for me so I can't duplicate the problem. I didin't
> find any problems with the model either.
>
> Try moving and/or rotating the model on the build plate. There is a bug
> that keeps good models from slicing when they are oriented certain ways in
> Cura and moving the model around can fool the bug.
>
> I'll mark this as a duplicate for the Cura team.
>
> [image: image]
> <https://user-images.githubusercontent.com/64202104/206310559-6e9e2e95-21dd-4c11-9d6b-7b870345efdf.png>

Hi,
I turn around the model and i was able to slice it.
Thanks a lot for your time!

Cassandra Lépine

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release

[image]

closed 2024-01-14 by jellespijker