Ultimaker/Cura #13989: Slicing Failed

DKiefer770, 2022-12-07:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

JGMaker ArtistD Pro

### Reproduction steps

Attempted to slice model.
Slicing Failed.

### Actual results

Slicing Failed with an unexpected error.  Please consider reporting a bug on our issue tracker.

### Expected results

Slicing should complete

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[Mark Hex 2.zip](https://github.com/Ultimaker/Cura/files/10179545/Mark.Hex.2.zip)
[cura.zip](https://github.com/Ultimaker/Cura/files/10179557/cura.zip)

GregValiant, 2022-12-07:
Thanks for the report.
There is a persistent bug that keeps models from slicing when they are in certain orientations on the build plate.
Try rotating and/or moving the model around.  I haven't been able to get it to slice though.  Cura hasn't crashed and I haven't gotten the "Make a Bug Report" message but it has been 45 minutes and it looks to be stuck about 25%.
BTW I think this model has been here before and I couldn't get it to slice then either.

I'll mark this as a duplicate and the Cura team will take a look.

DKiefer770, 2022-12-08:
Just FYI, this is the GCODE of a slightly different model that completed slicing, but the lower levels of infill look really strange.  I am going to try this again using cubic infill to see if that does any better.  (Actually the zip is 33M, so here is [a link](https://docautodev.sharefile.com/d-s50afffd2a8a3433b9a471c052cb9ff57).)

As a feature request, it would be great if under Gradual Infill Steps, there was an option to "Set Step Height Multiplier", which would default to 1 (which is the current behavior), so that if my step height was 3, and I set the Multiplier to 2, the topmost (smallest) fill layer would be 3mm, the second would be 6, the third would be 12, and so on.  Alternatively (or in addition to), a "Minimum Infill Density" would be great, so that if I have gradual infill layers turned on, I could set the least amount of infill to, e.g. 5%.  Ideally an "Auto Infill Gradient" would be cool, which would let the program calculate the number and size of the layers, from (e.g.) 35% at the top to 5% at the bottom.  This could be done in "Regular" steps of even height, "Multiplier" based step height, as mentioned above, or "Parabolic" steps that would compute the multiplier to gradually increase the layer size to maximize the thinnest infill layers to reduce the amount of material.  

Thoughts?

Thanks!