Ultimaker/Cura #13993: The problem of using brim and the full printable area of the printer for narrow bases

saf-yg, 2022-12-08:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Ghost 5

### Reproduction steps

[image]
Brim size affects print margins. Cannot use full printable area with brim for narrow bases. If you print without a brim, the model falls off.

### Actual results

If you change to resize the model, slicing of layers is not possible.
[image]

### Expected results

I would like that the size of the calculated model would be correctly taken into account or after the warning there would be an opportunity to continue working with slicing at your own risk.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

It seems to me that additional files are not needed to form an idea about this problem.

GregValiant, 2022-12-09:
Thanks for the report.
This has come up before and @Ghostkeeper did a nice writeup on why this is the behavior.  I was unable to find the reference but it boils down to...
When you bring any model onto the Cura build plate all Cura really knows are the dimensions of the bounding box that contains the model.  The measurements of the bounding box are X x Y x Z.  If the X and Y + build plate adhesion runs out of bounds then Cura won't slice.  The only way to determine that the features on Layer:0 are within the allowed area of the build plate are to slice the model and then do an analysis of the outside dimensions of the skirt/brim/raft.
There are workarounds for this that involve making the Cura build plate larger and then adding an M206 line to the very end of the StartUp Gcode to adjust the Home Offsets and move the print to the center of the printers real build plate.  An M206 should also be added to the beginning of the Ending Gcode to reset the Home Offset.

The Cura team will decide on this but right now what you see is expected behavior.  The last time this came up I think it was deferred.

Bzzz, 2024-11-12:
Probably related to #10166?
I just stumbled across this when scaling up a calibration cube for the printer of a friend that is said to have build plate adhesion issues. Just doing some silly 45°ish hollow tower thing as a simple benchmark of adhesion.
The scaled and tilted version slices fine with no or little brim, but once the brim is scaled too large, it fails slicing. According to 10166 (Cura 5.3), brims that exceed the build plate should just be discarded without error messages, since a partial large brim is still better than a smaller brim. Cura 5.8 however fails to slice.

30mm
[image]
40mm (same model size)
[image]

GregValiant, 2024-11-13:
Without a project file it's not possible to determine exactly why that happened.
This is in 5.8.0 with the print sequence "All at Once".
[image]

The only change here is to One at a Time.
[image]

Bzzz, 2024-11-13:
That is correct.
Running this sample with brim 10mm will slice, while with 20mm it won't.
Unless I set print sequence to all at once, then even a full-plate 300mm brim will do. Any partial brim is clipped at the edge of the print area as it should.

[cura13993.zip](https://github.com/user-attachments/files/17738654/cura13993.zip)

GregValiant, 2024-11-13:
The rules for One-at-a-Time are different as the machine head dimensions come into play.  If you are only printing a single part then set Cura to All-at-Once.
You can't print multiples of that test part in one-at-a-time so just set Cura to All-at-Once.

I'll remove the bug label from this and close it as what you are seeing is expected behavior given your settings.  I hope you understand.

closed 2024-11-13 by GregValiant

Bzzz, 2024-11-14:
I don't, since I would expect the edge cases "all 1 items at once" and  "the only one at a time" to yield the exact same results - but if that is how it works, well, go ahead. Nobody is talking about multiple items in that build volume, slicing fails with one item once the brim is enlarged. It is entirely non-obvious that this is due to a multi-part setting when no multi-part setup is even active.

Maybe a suitable hint can be added in the "unable to slice" notification, though.

GregValiant, 2024-11-14:
This is a reason I start every new project by re-selecting my custom profile and discarding all changes.  It's way to easy to forget that I've enabled Remove all Holes or moved the Zseam or turned on Z-hops for a specific model, etc.  The Print Sequence setting is just another example.  You can slice a single tall model in One-at-a-Time because the gantry height is ignored for a single model.  If it was not, then it would be a visible clue (on those machines with low gantry heights) that One at a Time is enabled.
But there is also "Bridge Settings" and "Support" which aren't required for every model and yet them might be turned on.