Ultimaker/Cura #13995: Once open a new window, the content cann't fill up the whole size

wasmann, 2022-12-08:
### Application Version

5.1

### Platform

windows 10

### Printer

Ultimaker S5

### Reproduction steps

Run the software and click a button to pop up a new window in the first time

### Actual results

[image]
The strange thing is that after rezising the window or dragging it to somewhere, then it turns to the normal size. And when I open the same window for the secound time, the issue will not happen again. 

### Expected results

make the window behaves correctly the first time to open it.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I am trying to make a fork of it. So maybe I changed something by mistake.

wasmann, 2022-12-19:
Hi, I cann't find the display settings in the screen shot as you said Display Settings or Scale and layout.

wasmann, 2022-12-27:
@MariMakes , Sorry I cann't  find the display settings in the screen shot as you said Display Settings or Scale and layout.

MariMakes, 2022-12-29:
Hey @wasmann,

I'm running on Windows 11, you seem to be running on Windows 10.
For you it probably looks like this
[image]

Here are instructions on how to adjust your scale
https://www.howto-connect.com/change-text-apps-size-windows-10-creators-update/

Can you let me know if this solves your problem, or if you need me to bring it up to the team?

wasmann, 2022-12-29:
Oh, I thought it was a setting in Ultimaker cura. 
I solved the problem by downscaling the display resolution to 100%, But the drawback of this approch is clear that all the icons and text become smaller which may cause discomfort to user like me. Is there another way to fix that?