Ultimaker/Cura #14001: Cura very, very, very slow, also 5.1.1, sorry to conclude afterwards

j-snijder, 2022-12-09:
### Application Version

5.1.1

### Platform

Windows 11

### Printer

Ender 3pro

### Reproduction steps

Standard working with programm, opening, loading stl, slicing, everything as normal work requires

### Actual results

Most of the time, if not all, Cura responds very, very, very, slow.

I had reported earlier that it did not occur with on my system windows 11 but actually it does, and not just a little bit.

As far as I'm concerned there is no satisfying way of working with Cura no more and I am sorry but if this is not solved I am going to abandon the, ever so good in the past, Cura slicer, and I guess I am not the only one that will leave.

Why is it so difficult to revert back to a version before v5 ( could it be the arachne engine ?, could it be the two latest Ultimaker printer models that causes all of this ?)

Blaming Windows 11 for all of this is in my opinion the wrong way to think, please keep us users informed and please, PLEASE, fix the many problems there are at this oment so we can enjoy our great hobby again, instead off just day after day frustrating about Cura.

### Expected results

Please fix it :(

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Please fix

MariMakes, 2022-12-13:
Hey @j-snijder,

I apologize if we are not informing you enough. 😕 
We are definitely working on it. 🤯 

The issue is indeed more difficult than we expected.
I personally upgraded my device to Windows 11 to help with troubleshooting but I'm not experiencing the slowness that you are describing. 

[We found this report in the community ](https://community.ultimaker.com/topic/42105-cura-521-work-slowly-and-freeze-when-i-go-to-the-printer-settings-menus-profiles-etc/#comment-315788) that it might be related to overlay software like Reverso or Punto Switcher that use global hotkeys. 

Can you check if you are having a similar program running in the background? I had Reverso enabled all day in the background yesterday and my Cura was noticeably slower, today I uninstalled the program and it seems to be running fine. 

Can you let me know if this might be the case for you as well?

j-snijder, 2022-12-13:
Hi,

I have no programms of that kind running, the only thing extra, harware wise, I have connected is a drawing tablet from VEIKK and a dial/toggle aid, the Logitec NuLooq, but that stuff is and was running perfectly together with Cura, no problems in the past whatsoever.

I hope you'll find the problem quickly because I miss out on options with the Creality slicer I am currently using to temporary replacing Cura.

Regards, Jan

j-snijder, 2022-12-13:
[image]

MariMakes, 2022-12-15:
Hey @j-snijder,

This issue is very much on our radar. I'll be closing the issue for administrative reasons. I'll close it as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/13820 

If you need any help or advice on specific models or 3D printing topics you are welcome to interact with the [UltiMaker community here](https://community.ultimaker.com/) instead of our issue tracking system. 

I hope you understand 🙏

closed 2022-12-15 by MariMakes