Ultimaker/Cura #14010: Failed to Slice

LiquidWayre, 2022-12-11:
### Application Version

5.1.0

### Platform

Windows 10

### Printer

Anycubic Vyper

### Reproduction steps

Clickled Slice Model, after 5 seconds, getting message about error during slice

### Actual results

No Gcode was produced, error issues in message box

### Expected results

.gcode file of the requested model

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[AI3M_GunStar_LastStarFighter.zip](https://github.com/Ultimaker/Cura/files/10201207/AI3M_GunStar_LastStarFighter.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/10201209/cura.log)

GregValiant, 2022-12-11:
Thanks for the report and the files.
There is a bug that can keep good models from slicing if they are in certain orientations on the build plate.  I moved your model and rotated it and it sliced.  Try that and see if it works as a temporary fix.

I'll mark this as a duplicate and a slicing error.  The Cura team will take a look.
Here is the sliced view.  With your settings it's nearly 45 hour print time.
[image]

LiquidWayre, 2022-12-11:
Moved model to position x:-20 by Y:40 without rotating the model and it successfully sliced.  
I had centered the model previously using the center model in Cura's menu.

jellespijker, 2024-01-14:
Thank you for reporting this issue, I can confirm that the project file still crashes with our latest internal build 5.7.0-alpha.0+3abb55 . For internal reference the crash is logged crash under Sentry issue: [CURAENGINE-1T](https://ultimaker-o7.sentry.io/issues/4859810080/)