Ultimaker/Cura #14012: Sizeable sections in top info bar

dlanders911, 2022-12-11:
### Is your feature request related to a problem?

I have a 32" monitor. I run Cura full screen. The info bar at the top, that displays View type, Color scheme, and profile sizes such that way too much space is given to Color scheme so that I can see any detail in the profile (See attached). 
[image]

### Describe the solution you'd like

I'd like to be able to size the sections using the dividers, as you would in Excel with columns, for example. Thus, I could see the print profile without having to expand the entire things, e.g., what is my infill percentage? That is one solution. 

### Describe alternatives you've considered

Any solution that would allow me to permanently set the size of each section, for example, percentages of display space, would be welcome.

### Affected users and/or printers

everyone with a large screen.

### Additional information & file uploads

Cura 5.2.1
[image]

fieldOfView, 2022-12-11:
The Sidebar GUI plugin removes the top bar and rearranges some other elements to solve this and some other issues. It is available in the Marketplace. You may like it.

dlanders911, 2022-12-20:
I am using the sidebar but it seems to slow Cura's response to mouse clicks. :( It also moves things around when you drop down the profile settings. Not really the solution I was looking for. 
Thanks, though, for the pointer.