Ultimaker/Cura #14013: Dialogs don't size correctly, and critical buttons are not accessible

dlanders911, 2022-12-11:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

Ender 3 Pro

### Reproduction steps

1. Every time I launch Cura and bring up any feature that presents a dialog box, the contents are not automatically arranged so that all of it can be seen.
2. Instead, I have to move to the edge and resize the dialog and then the contents are rearranged, and I can see things, like the buttons.

### Actual results

Examples attached
[image]
[image]
[image]
[image]

### Expected results

I should be able to see the buttons of every dialog box.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Didn't happen in Cura 5.0 (which I also have installed - is there a co-existence problem?) but has always happened in 5.2.1

MariMakes, 2022-12-13:
Hey @dlanders911,

Thanks for your report 👍

I think it might be related to having multiple monitors with different scaling issues. 
[Could you try the workaround mentioned in this community post? ](https://community.ultimaker.com/topic/41458-ultimaker-cura-510-window-scaling-problem/#comment-311838) 

If that doesn't work, let me know and I'll have someone from the team take a look 👍

GregValiant, 2026-09-27:
I'll go ahead and close this as being stale.

closed 2026-09-27 by GregValiant