Ultimaker/Cura #14014: Fail Slicing

dlock2, 2022-12-11:
### Application Version

5.2.1

### Platform

Windows 11 Pro

### Printer

Tronxy x5sa-pro 2e

### Reproduction steps

loaded a file from Thingiverse.  The download had 5 total files in it the other 4 sliced and printed fine.  

deleted from Cura opened again and failed to slice

deleted from Cura shut down Cura reopened loaded file failed to slice.

Deleted from Cura. Closed Cura, restarted computer, stated Cura loaded file failed to slice again.

Attached zip of .stl ans screen shot of error message

[MotorBeltTension_Ring.zip](https://github.com/Ultimaker/Cura/files/10201488/MotorBeltTension_Ring.zip)

[image]

### Actual results

The print failed to slice each time

### Expected results

The file should have sliced
[TXXPRO2E4_MotorBeltTension_Ring.zip](https://github.com/Ultimaker/Cura/files/10201495/TXXPRO2E4_MotorBeltTension_Ring.zip)
[log_file.zip](https://github.com/Ultimaker/Cura/files/10201496/log_file.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[TXXPRO2E4_MotorBeltTension_Ring.zip](https://github.com/Ultimaker/Cura/files/10201497/TXXPRO2E4_MotorBeltTension_Ring.zip)
[log_file.zip](https://github.com/Ultimaker/Cura/files/10201499/log_file.zip)
[image]

GregValiant, 2022-12-11:
Thanks for the report.
I think that model has been here before.  It has a lot of errors in it.
Report from Formware.co/onlinestlrepair
-> Analyzed your file:
--> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
--> 494 Non-manifold edges (?)
--> 3848 Inverted faces (?)
--> 632 Degenerate faces (?)
--> 16 Duplicate faces (?)
--> 0 Disjoint shells (?)
MS 3D Builder was able to repair the model and after repair it sliced with no problem.
If you load the Mesh Tools plugin from the Marketplace it will run a quick check on models when you open them in Cura.  That should inform you when a model has issues and you can repair it before working on the slicing.

I'm going to remove the bug label as this appears to be a problem with the model rather than with Cura.  I'll leave this open so you can respond and let me know if you feel that is premature.

This is the repaired model sliced using your settings (I just swapped out the repaired model for the original).
[image]

dlock2, 2022-12-11:
Sorry to bother you again.  I did the repair in 3d Builder and was not able
to slice.  I also installed mesh tools and the summary came back as
watertight.  I tried to fix .stl file at firmware.co and that stl would not
slice either.

I am getting the same error in the original bug post.  I have attached the
latest log file.

I really appreciate your help.  I am pretty new at 3D Printing.  I
appreciate your time and helping with my learning curve.

Darin