Ultimaker/Cura #14015: slicing failure

chaosmstr, 2022-12-11:
### Application Version

5.2.1

### Platform

Win10

### Printer

Creality cr6se

### Reproduction steps

downloaded https://www.thingiverse.com/thing:5625682
Edited part in meshmaker, size 2x, 2y, 1z, other edits.
exported STL.
Imported STL to Cura.
selected slice, got error after ~%50 processed.

### Actual results

cura gives error window.

### Expected results

should have produced gcode for sliced object.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

ZIP of files can be found here:
https://drive.google.com/file/d/11TvKRLyS-QAYmpdLUrFJL5eP6_W6-ISQ/view?usp=share_link

GregValiant, 2022-12-11:
Thanks for the report and the files.
I didn't find any errors in the model but it wouldn't slice where it was on the build plate.  After moving and rotating the model it sliced without issue.
I'll mark this as a Duplicate as this bug has been noted previously.  Hopefully it will be fixed for 5.3.
The Cura Team may close this as the behavior has been noted several times.
(I have no idea how I would get the support out of here.)
[image]

chaosmstr, 2022-12-12:
Thanks for the info. 
At least there's a work around for it! 
It had not occurred to me to simply move it around a little.
That fixed it.