Ultimaker/Cura #14020: Cura fails when trying to slice

timlocoweed, 2022-12-11:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

Ender 3V2

### Reproduction steps

1. Imported previously sliced and printed project file (yesterday): failed to slice.
2. Imported straight stl file that printed fine: failed to slice
[Card_HollowWing_w_Ring_Tight.zip](https://github.com/Ultimaker/Cura/files/10202926/Card_HollowWing_w_Ring_Tight.zip)

### Actual results

Receive generic slicing failed with an unexpected error message. 

### Expected results

Should of easily sliced.
[cura.log](https://github.com/Ultimaker/Cura/files/10202963/cura.log)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No other info

MariMakes, 2022-12-12:
Hey @timlocoweed,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

What a lovely model 🤩 
So festive and nerdy. 🤓 

I can see the crash in your logs: 
[MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [964]: Backend exited abnormally with return code 3221226505!
[MainThread] CuraEngineBackend.CuraEngineBackend._terminate [360]: Attempting to kill the engine process
[MainThread] CuraEngineBackend.CuraEngineBackend._terminate [366]: Killing engine process
CuraEngineBackend._terminate [369]: Engine process is killed. Received return code 3221226505

But when I try it on my Windows 11, Cura 5.2.1 I'm not getting the crash:
[image]

Can you check for me if you shared the correct model? 
Thank you 🙏

jellespijker, 2024-01-14:
Thank you for your report. It seems that your uploaded project file slices correct on our latest unreleased 5.7.0-alpha.0+3abb5 which indicates that the issue you're experiencing seems to be fixed for the upcoming 5.7.0 release.
I'm closing this issue as fixed for the 5.7.0 release

[image]

closed 2024-01-14 by jellespijker