Ultimaker/Cura #14021: Material sync never completes

Reynolds-Beta, 2022-12-11:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

Ultimaker S5

### Reproduction steps

1. Add a new material from the marketplace (in this case ESun PETG)
2. Restart Cura when prompted
3. Get the little pop-up dialog that says something along the lines of "A new material needs to be added". Click the Sync button.
4. When presented with the sync dialog, continue by pressing Start button

### Actual results

Sync never seems to complete. The dialog UI just says syncing and the spin control keeps spinning.

[image]

### Expected results

Confirmation that sync is complete, what was synced, and any errors.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Bonus points:
The icon for my printer seen in the attached photo shows a cloud icon. This printer is only connected locally to my internal network. I do not find any utility of using the digital factory so have not set it up. Shouldn't the icon be a check mark as in the main UI of Cura?

Sync UI:
[image]

Cura UI:
[image]

MariMakes, 2022-12-13:
Hey @Reynolds-Beta,

[I thought I recognize your name 😉 ](https://community.ultimaker.com/topic/41683-material-sync-not-working/?do=findComment&comment=315880) 
If you have already fount the community post, then I guess you have tried the troubleshooting steps as described here: https://support.makerbot.com/s/article/1667337559396

Can you share which material you are trying to sync?
The material sync can be quite finicky and can throw an error with the smallest inconvenience.

Reynolds-Beta, 2022-12-13:
> Hey @Reynolds-Beta,
>
> [I thought I recognize your name 😉 ](https://community.ultimaker.com/topic/41683-material-sync-not-working/?do=findComment&comment=315880) If you have already fount the [...]

I tried it with some custom materials I added to the materials list, as well as the ESun PETG from the Ultimaker marketplace.

[MHBuild material profiles.zip](https://github.com/Ultimaker/Cura/files/10220849/MHBuild.material.profiles.zip)

I only ever received the little pop-up about syncing with the marketplace material. Manually attempting to sync the customized materials (via material management) produces the same results.