Ultimaker/Cura #14022: Add Prusa Mini to pre-supported printers

Noxmiles, 2022-12-11:
### Is your feature request related to a problem?

I'm always frustrated with not having the Prusa Mini Profile for Cura

### Describe the solution you'd like

I believe this will spread love between the 2 biggest printer companies

### Describe alternatives you've considered

There is no alternative

### Affected users and/or printers

Prusa Mini, Prusa Mini+

Asterchades, 2022-12-12:
All 3rd party (non-Ultimaker) printer profiles are created by the community, not by Ultimaker. As such a pull request with an appropriate profile would need to be made to get them added.

There was such a pull request over two years ago but it was never finished. See https://github.com/Ultimaker/Cura/pull/7264

Not that there's anything stopping you from using any of the existing i3 profiles for the Mini, or any other profile for that matter, so long as you adjust the print volume dimensions appropriately. There's little if any model-specific information contained in any of them.

MariMakes, 2022-12-13:
Hey @Noxmiles,

@Asterchades is right! Since Cura is open source, you or anyone else is more than welcome to work on adding this 3d printer and create a pull request yourself. 

You can find instructions on how to add a new 3D printer definition here: https://github.com/Ultimaker/Cura/wiki/Adding-new-machine-profiles-to-Cura 
If you are stuck in any way, please reach out! I'll be happy to help out.

closed 2023-12-27 by github-actions[bot]