Ultimaker/Cura #14024: Slicer not working

MMCODER1, 2022-12-11:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Creality Ender-3

### Reproduction steps

I uploaded a .stl file, chose my settings, and clicked slice.

### Actual results

An error box appeared, saying "Slicing failed with an unexpected error." 

### Expected results

It should have sliced the .stl file.
[CE3E3V2_Powerful Leelo-Stantia (5).zip](https://github.com/Ultimaker/Cura/files/10203321/CE3E3V2_Powerful.Leelo-Stantia.5.zip)
[image]
[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

I can't find my log file, so please contact me if you can't help me without it.

GregValiant, 2022-12-12:
Thanks for the report.
The file wouldn't slice for me when I opened it.  Rotating the model 90° CCW fooled the bug and it sliced.
This is a duplicate of many other reports and the workaround of moving and/or rotating the model will often fool the bug.  Your model does have some errors that should be fixed before printing it.  MS 3D Builder is included with Windows and has a decent repair utility.

MariMakes, 2022-12-13:
Hey @MMCODER1,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

Thanks for the investigation @GregValiant!
[image]

This indeed seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/13643
We have a ticket on our backlog with the intent to improve this behavior.
I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/13643

closed 2022-12-13 by MariMakes