Ultimaker/Cura #14026: Slicing failed due to unexpected error.

MominNadeem, 2022-12-12:
### Application Version

5.2

### Platform

Window 10

### Printer

Ender-3 Pro

### Reproduction steps

I am working on a 3d project, with multiple files of an assembly.
all the files are working perfectly, but 1 file with gear is not slicing. I asked the designer to fix the file, but the error not due to the designer's fault. Cura is generating an error itself.

### Actual results

When I try to slice the file, an unexpected error screen appears.

### Expected results

I want the slicer to slice the file similar to the other files of this project.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

files.

MariMakes, 2022-12-12:
Hey @MominNadeem,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser. 

Thank you 🙏

MominNadeem, 2022-12-12:
[cura error.zip](https://github.com/Ultimaker/Cura/files/10207361/cura.error.zip)

MariMakes, 2022-12-12:
Hey @MominNadeem,

Thanks for sharing the file.
I'm also having trouble slicing the model in Cura 5.x.

I am, however, able to slice the model in Cura 4.13.
When I do I see all this texture on the outside of the model. 
[image]

I suspect that Cura 5.x is struggling with this texture. 
Is this texture intentional or do you think that the mesh is faulty?

GregValiant, 2022-12-12:
Hello to both of you.
@MariMakes I didn't have any trouble slicing the project in 5.2.1.  I exported the model and MS 3D Builder decided it had some errors.  The repaired model also sliced in 5.2.1.

I didn't see any "texturing" in either slice.
[image]

[GV_gear.zip](https://github.com/Ultimaker/Cura/files/10210293/GV_gear.zip)

MominNadeem, 2022-12-13:
thank you so much @MariMakes and @GregValiant for considering my request. This is good, that the problem is identified and solved. Let me pass my file through the 3D builder and see what happens.

closed 2022-12-13 by MominNadeem