Ultimaker/Cura #14031: Brim printed where prime tower would be, were there a prime tower to actually print

Reynolds-Beta, 2022-12-12:
### Application Version

5.2.1

### Platform

Windows 11

### Printer

Ultimaker S5

### Reproduction steps

Still getting used to the quirks with Cura, which I like a lot, but I keep finding inconsistencies. In this case, as I delve into printing composite materials on my UM S5, I find some of the settings behave weirdly.

1. Load a model (I used MatterHacker's astronaut)
2. Select Generic PLA as material, AA 0.4mm print core for extruder 1, PVA on BB 0.4 for extruder 2
3. Choose the canned "Default, Fast - 0.2mm" profile
4. Leave all other settings default (support OFF, adhesion ON)
5. Slice (Result: No prime tower even though both extruders are enabled)
6. Change extruder 1 to Generic CCF-CPE, CC 0.4 print core
7. Should already be set on the profile called "Resolution, Fast - 0.2mm" (I assume this is equivalent to 'Default')
8. Again, all other settings default (support OFF, adhesion ON)
9. Slice...

### Actual results

Slicer produces a brim around where the prime tower would be, yet unnecessary when the second extruder isn't being used.

### Expected results

No brim around a non-existent prime tower.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Attached project zipped files:

[UMS5_CCF_CPE_Astronaut Phil A Ment.zip](https://github.com/Ultimaker/Cura/files/10212215/UMS5_CCF_CPE_Astronaut.Phil.A.Ment.zip)

[UMS5_Generic_PLA_Astronaut Phil A Ment.zip](https://github.com/Ultimaker/Cura/files/10212217/UMS5_Generic_PLA_Astronaut.Phil.A.Ment.zip)

GregValiant, 2022-12-12:
Thanks for the report.
In the PLA file the Prime Tower is disabled so no prime tower = no brim.  If you turn Prime Tower on then the behavior is the same as CCF-CPE file.
In the CCF-CPE file the Prime Tower is enabled but has the Prime Tower Brim turned off so I would think it should not have a brim.  The brim appears to be carrying over from the model setting.

With the model Brim changed to Skirt and the Prime Tower brim disabled then the behavior is different but the skirt will still go around the prime tower location.  In that case to get the Prime Tower brim you would have to turn the Prime Tower Brim on.  It will show up even though it isn't being used.

Someone from the Cura team will take a look.
At first glance this would appear to be a bug, but really, you are asking for a prime tower so you are getting a prime tower.  The fact that it isn't being used may be over-ridden by the users request for a prime tower.  The question would remain why it's getting a brim when one is not specified in the Dual Extrusion settings.

Reynolds-Beta, 2022-12-13:
Ok, I guess I'm a little confused then: I never asked for the prime tower to be on, so switching to different materials shouldn't turn it on. Prime Towers in my humble opinion are optional constructs. Couple this with the expectation that a prime tower setting resides in the dual extrusion settings and no such dual extrusion is happening (only one extruder is being used in this print) means there shouldn't be a prime tower printed - or the brim around it.

Reynolds-Beta, 2022-12-13:
> I don't know how this came to be... The FAST profile for PLA _does not_ have the Prime Tower enabled. The FAST profile for CFF-CPE _has_ the Prime Tower enabled.

Well, inconsistencies tend to creep into any software - particularly in configuration - and is the bane of the software/service world. Probably just a cut/paste from some other profile that was missed during a PR and now has become part of history.

Thanks for the comments.

MariMakes, 2022-12-13:
Hey @Reynolds-Beta.

I checked a build with these changes https://github.com/Ultimaker/Cura/pull/11678/commits and it seems to automatically be solved there. So you can expect it to be fixed in the next Stable Cura release. 🎉

[image]

To me, it doesn't seem like a bug in Ultimaker Cura anymore, since there is a solution in place. 
Would you still like to keep this issue open or do you agree that we can close it?

Reynolds-Beta, 2022-12-13:
> To me, it doesn't seem like a bug in Ultimaker Cura anymore, since there is a solution in place. Would you still like to keep this issue open or do you agree that we can close it?

It is a _resolved_ bug until the aforementioned stable release happens and can be verified, after which it is appropriate to close.