Ultimaker/Cura #14032: failed slicing

Lancelotlink007, 2022-12-13:
### Application Version

5.2.1

### Platform

ubuntu 22.04

### Printer

Ender 3v2

### Reproduction steps

loaded model. sliced model @450%. failed.

### Actual results

failed to slice model

### Expected results

correct slice

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

nothing

jlreal, 2022-12-13:
I'm having the same issue on Windows 11 with an Anycubic Kobra Max

MariMakes, 2022-12-13:
Hey @Lancelotlink007,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser. 

Thank you 🙏 

@jlreal if you have a similar issue you can create a bug report here: https://github.com/Ultimaker/Cura/issues/new/choose

Lancelotlink007, 2022-12-14:
I figured out the issue. When slicing upscaled models with supports, do not
use tree supports. Once I changed the supports from Tree to Normal, the
issue was corrected.

MariMakes, 2022-12-14:
Thanks for the update. 
I'm happy your issue was resolved.
Let us know if you run into any other issues

I'll close this issue.