Ultimaker/Cura #14033: Prime tower brim when printing with single extruder

YoloDoge, 2022-12-13:
### Application Version

5.2.1

### Platform

Windows 10

### Printer

Ultimaker 3 Extended

### Reproduction steps

1. Enabled both extruders
2. Prepared a single model on the build plate
3. Selected "Extruder 1" to print the model with
4. Sliced the model

### Actual results

Model sliced **with** a prime tower brim

### Expected results

Model sliced **without** a prime tower brim. I expect that, when printing with a single extruder (even though that both extruders are enabled), the prime tower will automatically not be printed. As long as I can remeber, that was always the case in previous versions of Cura.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]

GregValiant, 2022-12-13:
Thanks for the report and the file.
There are a couple of things going on there.  Number 1 is that Cura should have refused to slice because the prime tower location is interfering with the model.
The second part (that the prime tower should not be there because it is unused) sounds like a duplicate of #14031.

That second part may or may not be considered a bug.  On the one hand, you have the Prime Tower turned on and you would be asking the software to over-ride a setting you made.  On the other hand, it isn't used so why is it there.  That might be because Cura doesn't know it won't be used until the slice is complete.  The slice might have to be performed twice in order to figure that out.
To avoid what is going on you could either disable Extruder 2 or turn off the Prime Tower.  Because of that I don't know how much traction part 2 will get with the Cura team.
But the first part looks like a bug to me as it really should not have sliced with the tower in the way because it definitely interferes with the model whether the prime tower is used or not.

YoloDoge, 2022-12-13:
As you have already mentioned in that thread, the FAST profile for CFF/CPE has standard Prime Tower enabled. I didn’t switch on that option.
When I re-slice the model, the prime tower brim remains. So, your theory about Cura not knowing that the prime tower won’t be used doesn’t seem to be true.
When I disable the second extruder, then the prime tower disappears (also the Dual Extrusion options are not available anymore, as expected).
I agree with you that the first part is a bug, but I think that (for the second part of this issue) that in the FAST profile for CFF/CPE “Enable Prime Tower” should be turned off, just like in the FAST profile for PLA.

GregValiant, 2022-12-13:
_"When I re-slice the model, the prime tower brim remains. So, your theory about Cura not knowing that the prime tower won’t be used doesn’t seem to be true."_
Ah, but you are assuming that Cura remembers the gcode from the first slice.  I'm pretty sure that isn't true.  So far as I know - every slice is unique and has no dependence on a previous slice and no bearing on the next slice.
I think I'll hang on to my theory until we hear back from the Cura team.

MariMakes, 2022-12-13:
Hey @YoloDoge,

Welcome to the Ultimaker Cura Github 🚀 

I recognize your issue, I make it myself aswell sometimes. 
I checked a build with these changes https://github.com/Ultimaker/Cura/pull/11678/commits and it seems to be solved there. So you can expect it to be resolved in the next Cura release 🎉 
[image]

Regarding the primetower enabled by default. 
We have print profile experts that tune the default profiles and they made a concious decision to keep the prime tower on for that material combination to improve the quality of the printjobs because both CFF and CPE are prone to stringing and a prime tower will reduce bleed in the dual extrusion printjobs. We are also planning to put some effort in making it more intuitive to select single extrusion for dual extusion printers so this prime tower pops up less. 

To me, it doesn't seem like a bug in Ultimaker Cura, since we have a solution in place. 
Would you still like to keep this issue open or do you agree that we can close it?

closed 2022-12-13 by MariMakes