Ultimaker/Cura #14036: Arc Overhang Support

mAd-DaWg, 2022-12-13:
### Is your feature request related to a problem?

Printing overhangs can be challenging and result in messy results.

### Describe the solution you'd like

Implement an option to print the first n layers of an overhang using an arc overhang algorithm, after which switch back to regular printing. This allows printing ridiculous without any support structures required, or if supported, with only tiny tower supports to anchor flexible edges to the build plate.

Links to the algorithm
https://github.com/stmcculloch/arc-overhang (original)
https://github.com/CNCKitchen/arc-overhang (cnckitchens improvements)

Link to a video demonstrating arc overhangs:
https://www.youtube.com/watch?v=B0yo-o47688

### Describe alternatives you've considered

Using various support methods. these are undesirable as they reduce the quality of the final print, can result in failures, and require a lot of tuning to manage t o get working decently. 

### Affected users and/or printers

It would effect all prints and printers, as the algorithm should be compatible with all 3d printers, and all overhang prints

### Additional information & file uploads

[image]

mm86133, 2022-12-14:
Voting for that too!

dominik59, 2022-12-14:
Please consider introducing it. It seems to be awesome ! :)

GregValiant, 2022-12-14:
I went to the Github site and downloaded a gcode file.  I pasted in my StartUp and Ending gcode and printed the file.

This was on my well tuned Ender 3 Pro in PLA.  The print time was 2:06.  It could have been faster as the base and strut could have been printed at 30mm/sec but the gcode file specified 10mm/sec and I didn't change it.  Layer Height is 0.4 and Line Width 0.8.  Print speed of the circle features was 2mm/sec.

[image]

[image]

There was about 3mm of warpage between the stanchion feature and the far point of the platform.  That will be hard to overcome as there isn't anything holding the far point in place in the Z.  There are gaps in the floor as well. The gaps may be a result of my printer not having been tuned for 2mm/sec extrusions.

It's an interesting approach but there are serious issues that need to be addressed.  For small delicate models, not having to remove supports would be a definite plus.

mAd-DaWg, 2022-12-14:
Interesting result. Did you use the cnckitchen one? they seemed to get pretty amazing results. Perhaps the gcode needs to be generated per printer config. i definitely see some spots where it looks like there was too much heat or more sensitive filament. the outer gap on the big arc looks like a flex break from the upper perimeter layers printed. perhaps adding anchors to the build plate on the outer edges would prevent that, or printing multiple arc layer before printing perimeters. i intend to experiment with this a bit and see what i can get out of it when i get time

GregValiant, 2022-12-15:
It was PLA at 200 so heat shouldn't have been an issue.  As the ledge built outward it was flexing and there is a slight curl that starts from the base of the first arc and sweeps upward until the tip of the end of the overhang.
I'm going to download a couple of the other files and see how it goes.
One problem may have been that I was too lazy to change to the 0.8 nozzle.  The velocity of the filament coming out of the .4 nozzle at .8 flow rates may have been an issue.  The warpage on the overhang will be hard to overcome.

GregValiant, 2022-12-17:
I've been trying hard to get this to work.

The overhang warps.  The warpage makes printing on top of the overhang dicey at best.
I made a part and utilized the "hard" gcode file from the Steve McCulloch site for the first layer of the overhang and specified two more layers above.  The Arc Support file was sliced at 0.4 Layer Height and 0.8 Line Width so I installed a 0.8 nozzle.
The print took a long time and finally finished with extremely poor results.
[image]

[image]

The floppy overhang did not allow the nozzle to exert any squish and so the filament just sort of laid there as round sausages.

For the next try I downloaded the CNC Kitchen "expert" gcode and created another model.  I added small supports in Cura hoping that eventually the overhang would grow over the support blocks and they would hold up the arc support overhang to provide some squish for the two skins that would follow.
Here is the model in Cura.
[image]

In post-process I cut out the first layer of the overhang and inserted the CNC kitchen gcode for the overhang.

Here the arc support is growing towards the conventional support blocks.  I thought I had something going at this point.  A bit later Sir Isaac Newton dropped an apple on my head.  The arc overhang started to fall below the supports and obviously wasn't going to finish so I aborted it.  Gravity won round two.

It would have taken about 4hr 15min for just the overhang to complete.  
For comparison - the print time for the conventionally supported model is 1hr 25min.

[image]

[image]

I've never found warts to be attractive.  Conventional supports on the right.
[image]

Gorlimtouk, 2023-01-24:
Very interested in a follow up and/or a Cura implementation.
I strongly believe that when implemented we should be able to maybe give small support pillars on only the largest circle-seed points / determend bridge-gap points to grid-form.
(to prevent the most mayor sags while also creating stability for slinger-beds / drastically minimizing support material, print time & gigantic overhangs.

The implementation of the CNC kitchen code to experimental would be very cool.
what notified me the most was his remark that the 2nd layer actually increased the 'droop' of the first layer; making the 2nd layer as important as the first arc layer for printing quality. 
Atm improving my ender 3 with upgrades and tuning it in. but once tuned in surely will look for updates and try to help with finding points/ideas for improving  on this arc-printing.

Orafilynie, 2023-03-13:
I hope this feature got added this year it could make me economize such a ton of filament and time :O

mAd-DaWg, 2023-03-19:
https://youtu.be/TGa_KvKLDR8

maruhe, 2023-05-30:
it seems like the only missing point is a comment to define the start of an overhang shape as SuperSlice / PrusaSlicer do when 'Detect Bridging Perimeters' is enabled
than this post-processing-script could be adapted to also work with Cura G-Code:
https://github.com/nicolai-wachenschwan/arc-overhang-prusaslicer-integration

maruhe, 2023-05-30:
[image]

maruhe, 2023-05-30:
[image]

tmbdev, 2023-07-23:
For me, this would be most useful for captive nut pockets and other kinds of smaller holes; those currently require a disproportionate amount of design and cleaning. It would be nice to just be able to put these into a part and not have to worry about them or the orientation that they are printed in. That way, one can focus on the bigger overhangs. Warping wouldn't be much of an issue for these since they tend to get attached within a layer or two anyway.

Lakeland97, 2023-09-28:
> [image]
> 
> This is indeed an interesting topic, just casually sharing an idea, what if these are combined with the tree support algorithms? What if we stop thinking about arc overhangs as a potential cure for the disease - supports; and rather start thinking about the possibilities of combining both techs to create something more reliable than both of them? We can prolly also utilize the best of the support interface tech out there.
> 
> Please refer to the above image, not trying to be artistic at all! If anyone gets the idea, likes it and want to implement it, I'll be glad to help in some way. Also, kudos to the OP for bringing up this topic 👍.
> 
> [Edit] Thinking about the 2mm or so warp that someone mentioned above, wouldn't the warp be beneficial for easily removing the supports?

turtle

vgdh, 2023-10-23:
is there any progress on this feature?

ozfunghi, 2023-11-02:
Cura team should be all over this. This is one of the 2 main reasons why i like to print with Fuzzy Skin in Cura, because in concept it works similarly for overhangs. Just not this extreme. And i think maybe the Fuzzy Skin algorithm could be used to expand upon. Fuzzy Skin also basically makes small outward overhangs, giving the next layer of overhang extra reach, especially combined with the Overhang Wall Angle/Speed feature (which was a feature request by me, 4 years ago, yay me). The rough texture of Fuzzy Skin helps to hide not just layer lines, but also drooping overhangs. 

Arc overhangs would take it to another level, and i hope to see this feature in Cura PRONTO!

JaneX8, 2024-01-03:
I would love to see this feature in Cura too.

Jan-Snijder, 2024-01-27:
If and when implemented I would choose for an extra option to tick in the bridge detection part, that way it makes more sense.

Just my 2 cents though :)

Sophist-UK, 2024-03-25:
Ok - here are my thoughts on how this might be implemented as a post processing script:

### Option 1
**Normal slice without supports and then convert normal slice to an arc-overhang slice in a post-processing script.**
1. Turn off supports. You may also need to change bridge settings, and that may create bridging issues elsewhere in your print. Slice normally. 
2. Post processor examines the Gcode, identifies where there are overhangs that need arc-overhang replacement and then replaces the existing gcode with arc-overhang gcode.

**Pros**
* Should use the same amount of filament overall.
* No changes to Cura needed

**Cons**
* Post processing complexity
* Potential bridging issues elsewhere

### Option 2
**Normal slice WITH supports and then remove supports and convert the normal slice to an arc-overhang slice in a post-processing script.**
1. Turn on supports. Other bridge/support settings may need to be carefully chosen. Slice normally. 
2. Post processor examines the Gcode, and identifies where there are supports / interface, which are removed (remembering the amount of extrusion that is saved) and remembering the areas that this covers. When it gets to bottom / bridging layer in areas where supports were removed, it then replaces the existing gcode with arc-overhang gcode, and adjusts the extrusion amounts as needed to account for the filament saved.

**Pros**
* No changes to Cura needed
* Manual supports where you still want to use them.

**Cons**
* Substantially more post processing complexity.
* Potential bridging issues elsewhere

### On second thoughts...
Writing a post processing script to re-work existing gcode is conceptually easier than writing a plugin, but it seems to me to be likely that the code to re-work gcode is likely to be substantially more complex than code to write arc-overhang gcode natively.

With Cura 5,5 Engine plugins, it is (in theory at least) possible to write a community plugin to generate arc-overhang gcode. This is also a significant and complex piece of code, but overall probably a little easier than the plugin. But it will need a hero who has the time and plugin experience to write it.

PeculiarDave67, 2024-05-17:
I just read about this technique, and I wonder if something similar could be done using the Mandelbrot set instead of Arcs for overhangs? 

No clue if this would be stronger, but how cool would it be to have **Mandelbrot Overhangs** in your design!

macckone, 2025-06-03:
Arcs are stronger than the classic mandelbrot set, the algorithm is a recursive set with curves.