Ultimaker/Cura #14041: (5.1.0) Supports begin in mid-air

uChip, 2022-12-13:
### Application Version

5.1.0

### Platform

Windows 10 home (19044.2251)

### Printer

Creality CR-10

### Reproduction steps

Seen on multiple .stl files, when slicing with supports turned on, sometimes support structures begin in mid-air.  They then make blobs that blemish the print.

### Actual results

see pictures

### Expected results

Supports should begin from a solid foundation.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]
[CCR10_Thick 4th Alternate gun.zip](https://github.com/Ultimaker/Cura/files/10222226/CCR10_Thick.4th.Alternate.gun.zip)

MariMakes, 2022-12-14:
Hey @uChip,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

[image]

That looks like a bug to me! 🐛 
I'll bring it up to the team to see what they can do to improve it. 
🤞 Fingers crossed

MariMakes, 2022-12-22:
We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-10065

Thanks for the report! 👍

DerGenaue, 2022-12-25:
FYI: Pretty sure this also is a duplicate of #12326