Ultimaker/Cura #14043: Cura 5.2.1 Slicing Pauses at 50%

ProtomakerSprint, 2022-12-14:
### Application Version

5.2.1

### Platform

Windows 11 64 bit

### Printer

Crealitit Ender 5 Plus

### Reproduction steps

Went to slice Cura 5.2.1 Slicing Pauses at 50% then after about 30 seconds to a minute continues slicing

### Actual results

Went to slice Cura 5.2.1 Slicing Pauses at 50% then after about 30 seconds to a minute continues slicing

### Expected results

Slice without the pause

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

#

GregValiant, 2022-12-14:
Thanks for the report.
Sometimes Cura needs to think hard about what it's running into in the model in order to correctly translate the model into gcode commands.  It doesn't "pause" but rather is "mulling something over" and processing the data.  I've noticed this myself on complicated models (and also on some models that I didn't think were so complicated but Cura did).
It's also a function of the combination of settings you have entered for the slice and of the resources available on your system as well as on the model itself and if other apps are getting processing / disc writing time.
I'll leave the bug label on this but I really don't see a bug.  Sometimes the computations take a while.  The variable line width of the 5.x versions of Cura do cause slices to take longer than in the 4.x and previous versions.

ProtomakerSprint, 2023-01-31:
Project file
[Use Touching Build Plate.zip](https://github.com/Ultimaker/Cura/files/10545843/Use.Touching.Build.Plate.zip)

ProtomakerSprint, 2023-01-31:
I do not know where or how to get a Log file?

GregValiant, 2023-01-31:
@ProtomakerSprint 
The instructions are on the first page you filled out for the report.
Create a project file by loading your model and setting up the slice.  Use the "File | Save Project" command to create a 3mf project file.
Use the "Help | Show Configuration File" to locate the Cura.log file on your computer.
Add both files to a zip folder and post the zip folder here.
An investigation depends on having both the Project File and the Cura log in order to tell what is going on within Cura.

closed 2023-02-15 by github-actions[bot]