Ultimaker/Cura #14044: [5.2.1] Program hangs/crashes on imporing STL with many polygons

franklinveaux, 2022-12-14:
### Application Version

5.2.1

### Platform

macOS 13.0

### Printer

Ender 3

### Reproduction steps

1. Import a file containing a large number of polygons.
2. Cura hangs forever with a frozen progress bar.

Note: File not uploaded because of Github limitations (file size 54 MB, Github rejects files over 25 MB because Github is laughably primitive and still stuck in 2003).

System: iMac 8-core core i9, 128GB RAM, 4TB SSD

### Actual results

Application hang/crash.

### Expected results

File should be placed.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

(File not uploaded because Github won't accept files >25mb)

MariMakes, 2022-12-14:
Hey @franklinveaux,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

If your model has a large number of polygons it might not be fit for 3D printing. Have you tried simplifying the file? for example the one on https://myminifactory.github.io/Fast-Quadric-Mesh-Simplification/.

Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser. 

Thank you 🙏

franklinveaux, 2022-12-14:
I do not have a project file because Cura crashes when I try to plac the model; it never gets as far as being able to save a project file.

MariMakes, 2022-12-15:
Hey @franklinveaux,

In your case, you can also share an STL file or a screenshot. 
But before you do, have you checked if your model is designed for 3D printing? 
Because if it's not it might not succeed to capture the detail or print small bridges. 

Did you try the simplification method I suggested?
Does that make a difference?

franklinveaux, 2022-12-16:
It is designed for 3D printing. I can’t share the model because it’s 54MB gzipped and the upload limit is 25MB.

MariMakes, 2022-12-16:
Hey @franklinveaux,

Thanks for getting back to us. I'm not sure how to help without more information about your print job. A screenshot would already really help. 

If you reach your size limit you can use a file-sharing site like WeTransfer and share the link in this thread.  https://wetransfer.com/

closed 2022-12-30 by github-actions[bot]