Ultimaker/Cura #14047: Material

TomEmmelot, 2022-12-14:
### Application Version

5.2.1

### Platform

windows 10 

### Printer

Vertex Delta K8800

### Reproduction steps

In "Material" I have set the "Print temperature of the first layer" 10 degrees higher, namely 240 as the rest of the print, but that does not happen, I do not find the code S240 in the GC code.
Declaration?

Kind regards Tom

### Actual results

No higher temperature.

### Expected results

Higher temperature.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[telefoon houder deksel.zip](https://github.com/Ultimaker/Cura/files/10228483/telefoon.houder.deksel.zip)

MariMakes, 2022-12-14:
Hey @TomEmmelot,

Welcome to the Ultimaker Cura Github 🚀 
Thanks for your report 👍

There are a number of settings that can influence your printing temperature. 
[image]

You shared an STL. Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

TomEmmelot, 2022-12-20:
I send it by mail also, reproduce the error more times now, option did not seems to work.

fieldOfView, 2023-01-31:
Cura has a "stacked" profile system; the actual value for a setting that is used can originate from different "sub-profiles". All settings can have a value for the printer type you use, for the material you use, for the quality, or for what you set in the settings floater/"side bar". Here is the technical background:
https://github.com/Ultimaker/Cura/wiki/Profiles-&-Settings

What you should know from all of this is that a value that is set in the quality profile, or in the custom print settings floater always overrides what is set in the material. That is by design, and unlikely to change any time soon.

What could be a UX improvement is if the Print Settings on the Materials pane of the preferences showed an indication if/when a value is overridden by the quality profile or a user value. 

The Material Settings plugin - available in the Marketplace - adds a "Use value from the material" entry to the context menu of the custom Print Settings floater. That will let you override what is set in the quality profile and revert back to what is set in the material.

TomEmmelot, 2023-02-01:
Hi @fieldOfView 

Can you tell me why than after the first layer there is a M104 but the S is the same as the start M104
The value stays the same, but the M104 is there!

Kind regards,
Tom